freifunk-gluon/gluon #429: Select GW with most announced bandwidth

RubenKelevra, 2015-07-31:
The currently behaviour of batman-adv is, as discussed some month ago in IRC and on the WCW-event, if there is a need to select a gateway (current gateway $GatewaySelectionClass points below the gateway with the most tq), select the gateway with the highes tq. If there are more than one gateway with the same tq, the first package which arrive will decide.

This tends to some gateways with more load and some with less. So the question was, how to let batman-adv decide better.

First approach is written on the very beginning of batman-adv, this has always been there, but has been always been broken: GatewaySelectionClass 1, which overflows an UINT32 in the latest version of batman-adv (2015.0). A fix has been commited here[1].

So we might want to change the GatewaySelectionClass to 1 per default.

Else I created a patch, we're testing it currently, which just select the fastest gateway from the current list of gateways, but filter out gateways which has a to small tq - means 20 points below the best gateway.

This patch is available here[2].

[1] http://git.open-mesh.org/batman-adv.git/commit/f63c54bba31d2c86269982fd8efdfb618f1daabe
[2] https://github.com/VfN-NRW/gluon/blob/stable/patches/packages/routing/0008-implement-selection-of-the-gateway-with-maximal-band.patch

Best regards 

Ruben

poldy79, 2015-08-05:
Currently, we decided to limit the number of VPN-Connections (fastd) to
one, as otherwise the overhead traffic is multiplied by the number of
active connections to GWs.

Perhaps there should more cleverness in fastd to connect to the currently
best GW and perhaps check from time to time if the decision is still valid.

Regards
Leonard

RubenKelevra, 2015-08-13:
I know that the most communitys use a 2 connection-limit for fastd, but the decision of the node, which gateway should be used is pseudo-random, leading to not-ideal load on some gateways.

Else the not-uplink-nodes might have more than two gateways availble with nearly identical TQs, in such cases the changes with my patch might help to equalize the bandwidth-usage in average on all gateways.

poldy79, 2015-08-16:
For commuities with a connection limit >1 your patch is definitively
useful.

We have currently a 1 connection limit to limit the background noise. We
have about 400 nodes within one L2 domain.

or having a wrap around script that reconnects (fastd) to the desired gw.
We also thougt about moving this logic onto the server, but currently fastd
lacks the ability to drop single connections.

Unfortunately we did not have time to dig into fastd.

Regards
Leonard

RubenKelevra, 2015-08-18:
Well, you might want to use dns-round-robin with one shared key for all gateways. Make sure you enforce a 1 second caching for the dns-entrys.

Else it is possible to answer with the server with mostly less load or traffic, with custom dns-scripts. But just heard about that - never used this.

christf, 2017-01-20:
in frankfurt we are doing this on the server side using this script: https://github.com/christf/fastd-peer-balance

is there a benefit of doing this on the client over the server-based approach?
If not, I guess this could be closed.

christf, 2017-01-20:
then again maybe both should be used to select the best gateway from 2 existing fastd connections...

jplitza, 2017-01-20:
In Bremen we try to steer nodes to gateways they have a good connection to (determined by their RDNS) by ignoring their first connection attempt: https://wiki.bremen.freifunk.net/Anleitungen/ProviderSteering

Regarding the original proposal, I think it is irresponsible to only take into account a value that the gateway operator has full control over, like the bandwidth setting. While using the TQ ensures that you can only locally damage the network as an attacker, relying on the bandwidth invites people to spawn gateways that are neither fast nor operated to serve common interests (and rather harvest login credentials). These gateways would then be selected by all nodes if their bandwidth is set to max.

RubenKelevra, 2017-02-16:
I guess the simplest solution would be to lower the chance of a successfully verify by a script on the server side, the more clients are connected and more gateways are online.

I don't see any benefit of a client-driven solution, since the client have no metric at all for this before any connection is established.

The only solution would be to ask each fastd-peer once for the client count and connect to the peer with the lowest count.

This would imply, that all gateways have the same available bandwidth and all nodes do the same amount of traffic.

mweinelt, 2020-03-02:
The gateway selection class could be exposed so communities can decide for themselves.

Excerpt from man batctl(8)
```

       [meshif <netdev>] gw_mode|gw [off|client|server] [sel_class|bandwidth]
              If no parameter is given the current gateway mode is displayed
              otherwise the parameter is used to set the gateway mode. The
              second (optional) argument specifies the selection class (if
              'client' was the first argument) or the gateway bandwidth (if
              'server' was the first argument). If the node is a server this
              parameter is used to inform other nodes in the network about
              this node's internet connection bandwidth. Just enter any number
              (optionally followed by "kbit" or "mbit") and the batman-adv
              module will propagate the entered value in the mesh. Use "/" to
              separate the down- and upload rates. You can omit the upload
              rate and the module will assume an upload of download / 5.
                        default: 10000 -> 10.0/2.0 MBit
                       examples:  5000 ->  5.0/1.0 MBit
                                  5000kbit
                                  5mbit
                                  5mbit/1024
                                  5mbit/1024kbit
                                  5mbit/1mbit
              If the node is a gateway client the parameter will decide which
              criteria to consider when the batman-adv module has to choose
              between different internet connections announced by the afore-
              mentioned servers.
              B.A.T.M.A.N. IV:
                        default: 20 -> late switch (TQ 20)
                       examples:  1 -> fast connection
                                       consider the gateway's advertised
                                       throughput as well as the link quality
                                       towards the gateway and stick with the
                                       selection until the gateway disappears
                                  2 -> stable connection
                                       chooses the gateway with the best link
                                       quality and sticks with it (ignore the
                                       advertised throughput)
                                  3 -> fast switch connection
                                       chooses the gateway with the best link
                                       quality but switches to another gateway
                                       as soon as a better one is found
                                 XX -> late switch connection
                                       chooses the gateway with the best link
                                       quality but switches to another gateway
                                       as soon as a better one is found which
                                       is at least XX TQ better than the cur-
```

NeoRaider, 2020-03-07:
`gw_sel_class` has existed as a site.conf option since 2015.

closed 2020-03-07 by NeoRaider