freifunk-gluon/gluon #432: Add documentation of supernodes

enko, 2015-08-05:
I have been reading gluon documentation, tickets and forums for the past few days, to get an idea about it and if it might be a good solution for our community. One thing that disturbs me a bit, is that some times there are mentions of „supernodes“, but I cant find any informations on these „supernodes“ in the [documentation](http://gluon.readthedocs.org/en/v2015.1.1/). I think it would be elemental to add the workings of a supernode to the documentation.

tcatm, 2015-08-05:
Where in the documentation did you find references to supernodes?

Most people build Freifunk networks using Gluon where Gluon is only a small part. Some of these people do document their setups. However, Gluon does not aim to provide a "reference setup" of any kind.

enko, 2015-08-05:
I wrote, I did not find any documentation of supernodes, even though they are mentioned in tickets (#298,#172) and a lot on forum.freifunk.net. And guessing from the name **super**node, this node-type is important?!?

After reading the [docs](http://gluon.readthedocs.org/en/v2015.1.1/), it seemed to me I only need to setup a fastd, a httpd for serving updates and creating a site specific firmware based on gluon to get started. But those mentioned supernodes made me a bit unsure.

tcatm, 2015-08-05:
It depends on what kind of network you are going to build. Freifunk-like networks rely on "supernodes" (it sounds way more important than it actually is) to provide routing to other networks (like the internet) and distribute IP addresses using DHCP.

enko, 2015-08-05:
Ah ok, but I can configure my nodes (a wireless router, for Example a TP-LINK TL-WR841N) in such a way, that they provide wireless access, route packets to the internet (if needed) and provide dhcp.

enko, 2015-08-05:
I think it is cleared what a supernode is and I will join IRC with my further questions :smile:.

closed 2015-08-05 by enko

A-Kasper, 2015-08-05:
Here are some links to some setups.. But there are many other ways...
1. https://pad.freifunk.net/p/r.YgRvmJ01ePukBuwG
2. https://www.freifunk-gera-greiz.de/wiki/-/wiki/Allgemein/Gateway
3. https://wiki.freifunk-franken.de/w/Freifunk-Gateway_aufsetzen
4. http://gluon-gateway-doku.readthedocs.org/de/latest/
You may use this as inspiration for your own setup.....