freifunk-gluon/gluon #433: Add a centralized node management interface

enko, 2015-08-05:
At the moment the autoupdater is dumb and will result in a broken network under some conditions. If you change the batman protocol, nodes will loose connectivity to neighbouring nodes and in result can't connect to the update server.

I propose to add a centralized management interface™. This interface will tracks the nodes version via gluon-alfred and manage the updates.

So if I want to release a new update, I would do the following steps in the interface:

* Add a new update
* Upload the image (Metadata like branch,version is extracted)
* Set a flag if the update needs a release.

If the update does not need a release, the behaviour would be the same as it is now. If you need to release the update for some reasons, all the nodes will download and keep the update. If you see it fit, that the update is ready, you can release it and the nodes will do the update.

A few unstructured thoughts I have in mind:

* What about nodes that are only sporadic online and are only connected via batman?
* I thought about a decentralized approach, but I think it would be to hard to implement for now. The data from gluon-alfred would need be in a decentralized persistent data structure. Also the node would need to figure when the update would be good to go. I think if someone wants to go down this path, it would be cool, but its too complicated and too many edge cases are ahead.
* I think it would be good to group nodes besides branches. So you can have a group with nodes that are only connected via batman and nodes that have wan access. Or maybe some other grouping I can't think of yet.
* The deployment of the management interface should be easy and have few dependencies. Maybe [ffmap-backend](https://github.com/ffnord/ffmap-backend) can be recycled for this.

Please note: This is a first draft and request for comments.

adlerweb, 2015-08-06:
Hmm... Centralized decision are always a good source for failures and/or possible security nightmares. However here my bits:
* Downloading the update beforehand will consume a lot of memory - not sure if it is a good idea to keep it there for longer periods of time.
* I would prefer staying away from ffmap-backend and alfred since they aren't exactly reliable

enko, 2015-08-06:
At Thu, 06 Aug 2015 00:33:57 -0700,
Florian Knodt wrote:
> Hmm... Centralized decision are always a good source for failures
> and/or possible security nightmares.

Of course, but a decentralized approach would require much more effort
and thinking. As for security: As I understand it, the updates are
already signed, so only updates with a correct signature will be accepted?

> * Downloading the update beforehand will consume a lot of memory - not
>   sure if it is a good idea to keep it there for longer periods of
>   time.

> * I would prefer staying away from ffmap-backend and alfred since they
>   aren't exactly reliable 

Thats good to know, since I'm relativly new to the hole gluon
stack. Knowing this, I think I would create a daemon which scans the
node for its version, saves the result to a database (what kind of
database is up to discussion, I preffer flatfiles for the moment,
since the volume of data is manageable) and also pushes
the update to the node via ssh.

I think since it might not be a good idea to save the update for a
longer time on the node, it might be good to calculate a update
priority score for all the nodes. For example: Nodes with only a batman link
get a high score, if the uptime isnt so stellar of the only batman
node, the score rises again. If the node has a wan interface, the
score will ne relativly low, since it does not depend on the batman
link.

enko, 2015-08-08:
I have updated the ticket to reflect the latest discussion on IRC and this Ticket.

NeoRaider, 2015-10-11:
Most Gluon-based network already have too much centralized infrastructure. We'd like to avoid by all means to add even more centralized control over the nodes... Giving one (or multiple) central server SSH access to nodes, even with the consent of the node owner, is not acceptable.

There are some ideas for decentralized management facilities though (see http://www.nilsschneider.net/2015/08/14/ff-online-config.html ).

Synchronizing firmware updates is a completely different issue, and there are some approaches to make it unnecessary (see #427 ).

closed 2015-10-11 by NeoRaider