freifunk-gluon/gluon #435: batman-adv-v15: double fragmentation issue (formerly: mow/mol issue)

tcatm, 2015-08-06:
batman-adv v15 does not correctly handle fragmentated packets when forwarding them through links requiring even more fragmentation, i.e. a 1500 byte packet is fragmented to fit within an MTU of 1400 and forwarded over a link with an MTU of 1300. In this case batman-adv v15 does not fragment the already fragmented packets to match the lower MTU but instead tries to send a too large packet through link causing that packet to be dropped.

Upstream ticket: http://www.open-mesh.org/issues/204

closed 2015-08-06 by NeoRaider

viisauksena, 2016-01-16:
i did not get this completly : for to clarify ...
did this happens with v14 also?
so i do decrease mtu of mesh-vpn to 1280 , or something?

jplitza, 2016-01-16:
No, this did not happen with v14.
Yes, 1280 is a common choice by many communities for the mesh-vpn. The rationale probably goes something like "It has to be fragmented anyway, so use something that will surely not cause any problems."

commit b408e1c9 (2015-08-06): batman-adv: decrease maximum fragment size

This should workaround batman-adv's inability to deal with different MTUs
which need fragmentation.

Fixes #435