freifunk-gluon/gluon #440: Add ability to limit bandwidth per client

Kunsi, 2015-08-16:
There should be an option to limit bandwidth for every connected client, so every client is able to get the full offered bandwidth, and noone is able to eat up all bandwith by transferring much data, which may result in other clients having no or very slow connection to freifunk networks.

Bandwidth limit for mesh clients (other nodes) should be configured by another setting, or set automatically (for example, five times client-limit)

MichaelSasser, 2015-08-26:
If there is a static limit, it is probably not longer possible to watch videos on Youtube etc. and  use the freifunk network to share data with other users (only verry slow).
If there is a dynamic limit, the node has to know the bandwidths of all nodes and balance the traffic perfectly through all other involved nodes and supernodes. Therefore the node has to distinguish the source and the destination of the traffic (through the exit VPN, through the gateway, inside lokal mesh and so on.). This requires too much cpu and ram. Traffic shaping needs some informations that are not stable as they are on a digital subscriber line.

Many devices are "sleeping" they are connected but only need less bandwidth. This topic is not easy.
We need a easy and clever idea to do that.

Joachim-Wolff, 2015-08-26:
Well, I think it make no sense to limit bandwidth between nodes, will make the network more or less unstable. If you don't have any bandwidth for other nodes, simply disable mesh0...

But it make sens to limit the clients, at least in some situations. 
But not sure it should be an option during the web based setup..

I'm meshing with a neighbor who provide/has the Internet access. 
The node-config is maybe of pre Gluon times with 2mbit Internet usage for the VPN tunnel.
Behind my nodes are 10-15 clients sometimes, so especial in the evening sometime nothing works any more. So I like to try a bandwidth limit to maybe 512kbit, so there need to to be at least 3 downloads at the same time to break down the line.

It is possible to shape/limit the bandwidth per client with the default installed tools like ebtables or iptables and on what interface? Can someone provide a sample config, please?

NeoRaider, 2015-08-26:
To do any meaningful traffic shaping, the bandwidth of all links must be known; for VPN links, it may be completely unknown. Also, the traffic must be shaped on all links for the shaping to have an effect.

Without shaping support from batman-adv, only traffic entering bat0 can be shaped at all, it is not possible to just shape the VPN tunnels as only batman-adv-encapsulated frames are visible there.

Furthermore, the correct bandwidth to limit a client to doesn't depend only on the information of a single node, we'd need to implement a daemon which exchanges traffic and bandwidth information with other nodes to get this right.

Searching for "traffic shaping" and "mesh" on Google Scholar yields a few papers with promising titles, but I have no idea if they are any good.

Joachim-Wolff, 2015-08-26:
Yes, you're right, it make no sense to drop an large answer packet just in front of the client, after it has passed all the bottlenecks of VPN and mesh... :-( And the upload is not the problem, most time...

I expect it will take some time to implement support into batman-adv, and also some additionally time to update an existing mesh to a new version...

The only workaround, but it's a really dirty one, is to shape to a "safe" value on the gateway, based/depending on the expected bandwidth to the originators MAC / node where the client is connected to. Don't know its possible, but maybe it helps a little bit for nodes with 30+ clients and not so fast Internet connections, if this may be an issue...

RubenKelevra, 2015-11-09:
I think trafficshaping within the mesh-network not linkbased, like vpn-traffic-shaping but for some stations is against the idea of the pico-peering agreement.

CodeFetch, 2019-03-18:
Here are my points and the reason why this issue should be closed. Traffic shaping on a downlink (what you seem to aim at) works (only) due to fq_codel (because you can't force a server to throttle its uplink without it). That means you could, but it should not be implemented in Batman-adv. Whatever you do on the downlink to throttle the traffic, if its not something like fq_codel the VPN link might still be congested for nuts.
There are two technically completely different scenarios of what you want to achieve:
- You want to throttle a client's traffic to a hardcoded limit (like mobile network providers - e.g. only 56 kBit/s)
- You want to achieve fairness (that no client uses all the bandwidth and the other clients have problems)

The solution for the first scenario is viable by implementing rate-limiting in Batman or much easier and what providers actually do, put each client into a separate VLAN and use tc for rate-limiting.
I assume as you are no mobile network providers who want to cut costs there are only two problems why you want such a solution:
You don't have enough VPN bandwidth and clients on single nodes with too much traffic disturb the network. This could only happen if you use tunneldigger or something. If it happens with fastd, you should really rethink your infrastructure requirements. The solution is to rate-limit the nodes VPN using tc. Actually this should never happen as it means that your infrastructure is undersized.

The reason why I think most of you want this "feature" is not that you want to throttle clients, but you want to achieve fairness. The solution for this is airtime fairness and it has been finalized last year in the linux kernel and is part of Gluon v2018.2.
Thus I think this issue is resolved now.

closed 2019-03-18 by rotanid