freifunk-gluon/gluon #442: Support for Netgear WNDR3700v5

mmalte, 2015-08-21:
Hello,
we are running short on any fully compatible gluon compatible devices with dual band support.

I'd love to see support for the Netgear WNDR3700v5:

http://wiki.openwrt.org/toh/netgear/wndr3700
https://forum.openwrt.org/viewtopic.php?id=56737&p=2

Is it reasonable to order one to investigate the possibilities?

NeoRaider, 2015-08-21:
At the moment, there is no working driver for MediaTek's 5GHz WLAN (at least not on the Archer C2*, which use a MediaTek chipset as well).

Also note that the Ralink-based MediaTek WLAN adapters will never support AP+IBSS mode, but as soon as IBSS has been replaced with 11s (as already supported by the Gluon master), we'll finally be able to support some non-Atheros chips, including Ralink-based ones.

mmalte, 2015-08-21:
Okay, so this is an openWRT and not an gluon issue.

Are you able to guess how long it usually takes to solve such an issue.

What does it take to support the community to get this done?

Do you recommend to work on this device or is there another more promising dual band device for around 50€?

NeoRaider, 2015-08-21:
The time to get WLAN driver support for new chips? Can range from months to decades... Although the development of the MediaTek drivers is more or less active, so having them supported this or next year is likely. In contrast to the basic platform support, which is realized a a Linux kernel patch and thus must be open due the the terms of the GPL, the WLAN drivers are usually complely closed, so unless the vendor supports the development of open drivers, reversing them is impossible for most people...

At the moment, I don't know of any devices I could recommend. Most new dualband devices have 802.11ac, and as far as I know, no 11ac-cabable card has good open source drivers yet. I think it'll take a few more years until 802.11ac is really usable with open software, but it is getting harder and harder to find old (ath9k-based) dualband hardware...

mmalte, 2015-08-21:
Well, that sounds like I'll focus on the archer c5 because there is nothing I can do about this one.

commit 3ec63b8c (2018-01-19): ar71xx-generic: add support for Netgear WNDR3700v5

fixes #442
closed 2018-01-19 by rotanid