freifunk-gluon/gluon #444: trim hostname in lucy input field

rubo77, 2015-08-22:
When entering a hostname starting or ending with a space, you get an error message, that there are invalid characters, but it is not really visible.

same with the coordinate fields.

It would be handy if all field entries would automatically be trimmed on save so you only get an error message if you do something really wrong.

tcatm, 2015-08-22:
Could you contact upstream (OpenWrt) to check whether it can be fixed there?

rubo77, 2015-08-24:
I thought this would be something the lua scripts would have to manage? Who is creating the error message that there are false characters in the hostname?

tcatm, 2015-08-29:
Verification is handled by luci in this file https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/cbi/datatypes.lua

rubo77, 2015-08-30:
https://github.com/openwrt/luci/issues/458

rubo77, 2016-07-10:
This could also be handled according to this package: https://github.com/ffgtso/ffgt_packages-v2015.2/blob/master/gluon-config-mode-hostname-ffgt/files/lib/gluon/config-mode/wizard/0100-hostname.lua

NeoRaider, 2017-02-23:
Obsoleted by pretty-hostname support.

closed 2017-02-23 by NeoRaider