freifunk-gluon/gluon #446: Load peak every 42 minutes

Joachim-Wolff, 2015-08-26:
I've observed some load peeks some time ago, in v2015.1 also present in v2014.4, sometimes up to  a load of 5. During this times the packet forwarding is hardly affected, at least over WLAN.

Now I find out they occurring regular every 42 minutes, first after approx 10 min uptime. After a node reboot they occur at different times, depending on the uptime, so they can't be triggered by outside effects. Seams to be independent from number of clients or mesh link quality.

In Hamburg the average peak is a load of 2.0 while at Pinneberg community (with less nodes) only at 0.75

Is there maybe a clean up job in the kernel or batman that is responsible for this issue?
The impact to nodes within mesh environments is maybe much bigger than for standalone only VPN connected nodes.

[image]

Joachim-Wolff, 2015-08-26:
Also nodes without clients and disabled mesh0, only connected over br-wan, are affected in the same way. (ZwergenAufstand-Test)

[image]

NeoRaider, 2015-08-28:
This is really weird. My only idea is that some counter wraps every 42 minutes, presumably in batman-adv. Which compat version are you using?

Can you see the name of the threads causing the load, maybe using a nicer top tool like htop? It can be configured to show kernel threads as well (I think it even does by default, but I'm not sure).

NeoRaider, 2015-08-28:
Please also check if this still occurs with the current Gluon master (or the slightly older commit 11bffa59db923e976587cfa981ee4d95d1307e5b, which still uses the old WLAN configuration, so you don't have do adjust your `site.conf`).

Joachim-Wolff, 2015-08-29:
The nodes are in Hamburg, so it's batman adv 2013.4.0

I've updated one node to v2015.1-70-ga581477 two days ago, so it based on Chaos Calmer, but it shows the same behavior. 

I've seen up to 38% sirq just in front of such a load peek, while got only 1,4% user, 0,5% sys and
ksoftirqd/0 was at 0.9% cpu displayed. No idea where the 38% coming from... Was nearly no client traffic at this time.

RubenKelevra, 2015-09-03:
Might be some Broadcast-Mesh-Bursts ... did you tried to monitor the network-traffic, if there are any peaks on the wan-port?

bittorf, 2015-09-04:
42!!! 8-)

Joachim-Wolff, 2015-09-04:
Yes, have tried to capture traffic via tcpdump on one node, but resulting in very high load.. :/
Will get a cisco switch where I can configure a monitor port, hopefully in 3 weeks, so I can look on the br-wan interface without any impact to the node.

I don't expect it has something to do with external traffic, because then the time when one node is affected will not be changed by a reboot of it.
And with broadcast traffic all nodes should be affected at the same time, theoretically...

mmalte, 2015-10-10:
I'm able to confirm this issue:
https://forum.freifunk.net/t/load-spitzen-in-regio-aachen-und-rheinufer/5900

mmalte, 2015-10-10:
Here is another example: The four nodes running in the Regio Aachen.

Green and blue conected via Wifi-Mesh
Red and purple connected via Wifi-Mesh

[image]

Adorfer, 2015-10-29:
i can see a link to packet-rate bursts on the in background-traffic (not forwarding, rx or tx) 
this causes the load of the node, even if it's "wifimesh-only" and "0 clients". Even a WR3600 will hit the 1.x load. 
(Batman adv15, 150 nodes, 300 clients in domain)

Joachim-Wolff, 2015-10-30:
Well, I've captured some hours of traffic like described earlier, but not finished analyses completely yet..

There are some traffic peaks caused by ipv6 Multicast Listener Report Message v2 (143) exactly every minute, synchronously transmitted by all routers and some peaks of ipv4 ARP caused by port scans in my case...

But this will not resulting in a load peak every time, but near to a load peak this traffic condition seams to be present. Will invest some time next week to review the captured data...

[image]

Don't worry about the packet counts in the graph, every broadcast packet is captured 4 times in this setup.

Joachim-Wolff, 2015-11-10:
Don't find anything more related to the load-peaks in the captured traffic...

mmalte, 2016-02-24:
I do no longer see this load spikes, nodes are running on 2016.1

[image]

Adorfer, 2016-02-24:
@mmalte 
Dito hier: 

http://i.imgur.com/wJgTnWt.png

NeoRaider, 2016-02-24:
Great, closing this ticket.

closed 2016-02-24 by NeoRaider

tobox, 2016-09-26:
Load Spikes are back in 2016.2 at FFRN, again at 42 minutes:

http://s.ffrn.de/dashboard/db/nodes?panelId=3&fullscreen&var-Node=Bensheim-002&var-Node=Bensheim-006&var-Node=Bensheim-014&var-Node=Bensheim-015&from=1474789943446&to=1474817711147

T-X, 2016-09-26:
@tobox: Are you using 2016.2 with or without the new gluon-ebtables-segment-mld package?

tobox, 2016-09-26:
We are using this site.mk:

https://github.com/Freifunk-Rhein-Neckar/site-ffrn/blob/ffrn-v.0.6.x/site.mk

So yes, it should contains gluon-ebtables-segment-mld.

T-X, 2016-09-26:
@tobox: Ok, thanks! Are the peaks gone again on the Bensheim nodes when you switch just these nodes back to the older, previous firmware?

(to know whether we need to look for the cause on the local nodes or whether something somewhere else in the mesh is causing it)

tobox, 2016-09-28:
@T-X I did not have time to downgrade my nodes. But may nodes in out network are still running a firmware based on 2016.1.6, and they do not have any load spikes, e.g.:

http://s.ffrn.de/dashboard/db/nodes?panelId=3&fullscreen&var-Node=Bensheim-001