freifunk-gluon/gluon #447: Announce client count separated by frequency

jplitza, 2015-08-27:
It might be nice to distinguish how many clients are connected to a node via 2.4 GHz and 5 GHz, so we could gather information about the usage of the 5 GHz band if available.

While the actual implementation is rather easy (we already used [a hackish version](https://github.com/FreifunkBremen/ffhb-packages/blob/breminale/ffhb/ffhb-breminale/files/lib/gluon/announce/statistics.d/clients) that simply parses the output of `iw dev ... station dump`), we should probably first fix the output format in the JSON. Currently, we have:

    "clients": {
      "wifi": 3,
      "total": 3
    },

I would propose the following format, which obviously isn't backwards compatible:

    "clients": {
      "wifi24": 2,
      "wifi50": 1,
      "other": 0
    },

Having a "total" field doesn't make much sense when having more than one other field, I think, because you can get the total by something like `.clients | add` (in jq), whereas it would be more cumbersome and less flexible to get the count of "other" given "wifi24", "wifi50" and "total".

(I was by the way told that @NeoRaider already approved of the general idea at CCCamp, and that he would like different types of clients to have different color tints on the map)

NeoRaider, 2015-08-27:
I like the general idea of differentiating the clients by frequency. I never said anything about client colors, but I guess that makes sense as well. Instead of using the `iw dev` commands, obviously iwinfo-lua should be used.

I'm not sure about the exact JSON format... we could just add the `wifi24` and `wifi5` (`wifi50` doesn't make any sense, the 5GHz band goes up to 5.8GHz) fields and keep the current `wifi` field for a while to maintain some backwards compatiblity. I don't think we should replace `total` with an `other` field.

tcatm, 2015-08-27:
Does NetJSON have a solution for this?