freifunk-gluon/gluon #451: status-page & wifi-iface mesh_radio0 disable

Joachim-Wolff, 2015-08-29:
If wifi-iface mesh_radio0 is disabled in /etc/config/wireless, the status-page show the usage of "iw dev"

Putting a # in the system call helps as dirty workaround, /lib/gluon/status-page/www/cgi-bin/status line 79... 

At least v2015.1 + v2015.1.2 are affected.

tcatm, 2015-08-29:
Thanks for pointing this out. The "old" statuspage was never meant to be perfect. It's just a few hardcoded commands that would output useful information in most cases and it will be replaced with the a new status page soon.

kb-light, 2015-08-30:
Should already be solved in the current Master since 753c1f9, since iw dev <dev> link is not used anymore in the "old" statuspage.

NeoRaider, 2016-02-05:
As @kb-light pointed out, this has been solved already.

closed 2016-02-05 by NeoRaider