freifunk-gluon/gluon #452: Get rid of rpcd?

NeoRaider, 2015-08-29:
Before making a Chaos-Calmer-based release we should look into getting rid of rpcd, which is a completely unneeded dependency of LuCI (it is used for authentication only in LuCI 1, but Gluon doesn't use authentication in LuCI). While rpcd is very small, having an unused daemon runing all the time doesn't seem like a good idea.

FreifunkUFO, 2015-09-04:
rpcd is needed for any ajax/javascript based webservice, for luci2 (a very small webservice) and also for getting statusinformation/ setting configuration via http.. it connects via ubus.
(f.i. alfred values can be read via rpcd: https://github.com/freifunk-leipzig/solarfestival-packages/blob/master/rpcd-mod-alfred/rpcd/usr/libexec/rpcd/alfred )

it seems that gluon-firmware has done that some months ago via an extra new service called "gluon-announced". Now that could be migrated to rpcd now.

tcatm, 2015-09-04:
Both rpcd and gluon-announce serve different purposes. rpcd can not be used in place of announced (and announced can not replace rpcd).

commit a71ab2fb (2015-09-05): Get rid of rpcd

LuCI's authentication won't work without rpcd, but we aren't using the
authentication anyways. Users who need it can just install rpcd explicitly.

Fixes #452
closed 2015-09-05 by NeoRaider