freifunk-gluon/gluon #453: Change label for altitude

rubo77, 2015-08-29:
This is a followup of #381:

> we should have a button: "show altitude of streetlevel" in the mesh viewer or a link to a map, that has this easily accessible

Or we just change the label to 

"Altitude above streetlevel"

do9xe, 2015-09-13:
Is this a request for a change in the packages textfiles or a request to add a feature to the mesh-viewer? If it is for the meshviewer, you are in the wrong repository.
If it is for the package: please create a pullrequest by yourself.

do9xe, 2015-09-13:
After talking to @jplitza and @tcatm we decided that the visualisation of the altitude gets to complicated, when you use the altitude over ground. So this is going to stay altitude over NN, until we discussed this in a propper way.

rubo77, 2015-09-13:
Until now its not visible that it should be "altitude over NN" it just said "altitude"

Did you find a map that shows the right altitude?

jplitza, 2015-09-14:
As mentioned yesterday, we should define more precisely what is meant by "latitude" and "longitude" (as in "according to WGS84 projection") as well.

rubo77, 2015-09-27:
NN is replaced by [NHN (Normalhöhennull)](https://de.wikipedia.org/wiki/Normalh%C3%B6hennull#Umstellung_von_Normalnull_auf_Normalh.C3.B6hennull)

so the title for https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-config-mode-geo-location/i18n/de.po
should be

    msgstr "Höhe über NHN"

with NHN as an external link to get the height for the entered coordinates in the fields above

bittorf, 2015-09-27:
rubo77: 8-))) *no*body (except you) knows NHN without explanation

rubo77, 2015-09-27:
The myin point is, We should use the most commonly used system and add a link to a site te receive that altitude in config mode directly, so everyone uses the same system.

(by the way NHN only differs by some 10cm from the old NN)

bittorf, 2015-09-27:
rubo77: now i'am unsure: do we talk about "Altitude above streetlevel" or NN? also: better use 'NN' if you really talk about this: it's difficult anyway to measure that exactly...

rubo77, 2015-09-27:
We could get the information from the Google Elevation API.

By making a GET call and passing in the latitude and longitude in the query, you can get back the altitude from sea level in meters:

    https://maps.googleapis.com/maps/api/elevation/json?locations=40.714728,-73.998672

This will return a json object like this:

    {
       "results" : [
          {
             "elevation" : 8.883694648742676,
             "location" : {
                "lat" : 40.714728,
                "lng" : -73.998672
             },
             "resolution" : 76.35161590576172
          }
       ],
       "status" : "OK"
    }

More info: https://developers.google.com/maps/documentation/elevation/intro

*Source: http://stackoverflow.com/a/31839572/1069083*

Adorfer, 2015-10-30:
even it's transferred from NN (depreciated unit) to NHN (correct term): 
If you ask for NHN, people will start using their smartphones! This is ruining everything!

why? the accuracy of GNS for z axis is by design around 1/10 of the lat/lon.
So if you have basic accuracy of +/-3m (by the use of WAAS), your accuracy in altitude will bei +/-30m. 
This might not be fatal if you are living a mountain area. 
But in basically flat area where everything is e.g. around 100m, you will end up with lots of nodes at 80m besides a node on 125m... on the same flat street with small houses.

If you would just take a value "hight over ground, calculate one level as 3m": It would be rough, but at least more accurate for visualization.

rubo77, 2015-12-04:
rebased PR: #581

closed 2016-12-11 by NeoRaider