freifunk-gluon/gluon #454: RFC: Decentralized firmware mirrors

jplitza, 2015-08-30:
Up until now we only had one firmware mirror that was used by the autoupdater. To remedy this situation, I came up with the idea that node owners with fat internet connections could simply check a box in the config mode, attach a USB storage device, and the node would take care of everything else.

While fetching the updates and keeping them up to date probably wouldn't be big of a problem (I will have a go at this soon – hopefully), it still would take a firmware update each time someone sets up a new mirror in order to let the other nodes use it.

So I thought that maybe we could decentralize this process:
* Every node may act as a firmware mirror if the owner wishes so. Keeping the files up to date could happen for example via a cronjob with wget.
* Every mirror announces that it is a mirror having this and that version available, for example via our usual announce mechanism.
* A node looking for an update then queries who is a mirror (e.g. by using respondd) and selects a mirror with the newest available version.
* If something goes wrong using that mirror, it tries the next one, even if it announced an older version.

What do you think of this? I'm still undecided whether an updating node should try the mirrors in its site.conf first or last, or maybe a mixture (try it first for the manifest and last for the images).

Also, in theory, nothing restricts this idea to nodes: "Real" machines could also act as firmware mirrors in this way.

tcatm, 2015-08-30:
What are the problems solved by this? Can this be solved by implementing a "decentralized webserver" so we can keep the autoupdater unchanged and unaware of the underlying network structure?

jplitza, 2015-08-30:
The "problem" solved is that currently, being a firmware mirror is something "exclusive": Only a few URLs are present in the site.conf, and every time someone sets up a mirror a new firmware release has to be done to update the list of URLs on the nodes.

Another "problem" solved by the selection of the "newest" mirror is that of a need shared webspace. Everybody who can build firmwares (i.e. have their pubkeys in the site.conf) currently also has to have access to all mirror servers in the site.conf to update the firmware there (or if they mirror automatically, to the primary source server). With this change, everybody who can build firmwares can set up his or her own mirror and publish the firmware there - it would automatically be preferred, because its the newest.

However, I thought of this less as a "problem solver" and more as a "feature".

And not, this cannot be solved by a "decentralized webserver", because of the selection of the mirror based on the newest version.

RubenKelevra, 2015-09-03:
Well you might want to add an .local DNS-Name, which would be resolved automatically by all nodes :)

Than you can add the hostname to an avahi-daemon which should be started, if you got images on a stick, and than all nodes can update from this stick. :)

bittorf, 2015-09-04:
why cant you use just an anycast IP like 192.168.254.254 for the update server. every node which want to be an update server just has this IP and every router has an dns-entry where updateserver.local points to $IP

NeoRaider, 2015-09-04:
@bittorf, it's not that simple in a layer 2 mesh network. As our nodes are only bridges and not routers, we can't use IP anycast. @tcatm is working on integrating BABEL into Gluon, though...

bittorf, 2015-09-04:
@NeoRaider this is not true. just add this IP to some of the nodes and try to ping it. the nearest works...

NeoRaider, 2015-09-04:
Please note that we won't add IPv4 support to the nodes, and for IPv6 the DAD will definitely break this (unless we disable DAD, which doesn't seem like a good idea either).

bittorf, 2015-09-04:
i think there is support for anycast IP's (IPv4 + IPv6) in newer batman-adv versions.

jplitza, 2015-09-04:
@bittorf Also, we would have no control over which mirror to choose. If the one anycast address doesn't work, we cannot use another even if plenty are there. And choosing the newest wouldn't be possible either.

NeoRaider, 2015-09-04:
I think what batman-adv implements is Ethernet anycast, which can then be used to implement IP anycast. For this Ethernet anycast, we'd at least need to add another macvlan to br-client. (See http://www.open-mesh.org/projects/open-mesh/wiki/Connecting-Batman-adv-clouds#Layer-2-Anycast )

I don't think we want to add any more batman-adv-specific features to Gluon for now, as we're trying to get rid of it (or at least make the routing protocol configurable).

bittorf, 2015-09-04:
@NeoRaider you are right. but you can workaround this be search the "real" IP of all these nodes.
than try iterate over the list to download it from the real IP.