freifunk-gluon/gluon #458: Feature Request: zusätzlichen personalisierte SSID

A-Kasper, 2015-08-30:
Hi
viele Unternehmer, die sich am Freifunk beteiligen fragen, ob es möglich wäre den Netznamen irgendwie an ihren Betrieb anzupassen. Ich fänd es gut es zu ermöglichen, dass man via Lucy eine ZUSÄTZLICHE SSID ins Freifunk-Netz aktivieren kann, die  Community SSID soll hierbei wie gehabt bestehen bleiben. Ich könnte mir vorstellen hierdurch noch weitere Mitstreiter gewinnen zu können, weil diese auf diesem Weg für sich werben können.
Grüße André

corny, 2015-08-30:
Und wie könnte so eine zusätzliche SSID ausstehen? `Dieser Freifunk-Node wird gesponsert von ...?`. Ich halte persönlich nichts von dem Vorschlag.

A-Kasper, 2015-08-30:
In der Regel wollen die sowas wie Kneipenname-Net 
Es wäre halt zusätzlich. Wer es braucht könnte es machen. Ist nur Addon und tut nicht weh. Ich habe nix dagegen, wenn jemand der einen Freifunk-Knoten aufstellt dies an die große Glocke hängen mag. Ist immer ein nehmen und geben. Communities könnten ja selbst entscheiden ob sie ein entsprechendes Paket einbinden oder nicht

Kunsi, 2015-08-30:
Nächste Stufe ist dann "aber ich will das limimiteren, und ne Splashpage dazu haben, und …". Find ich nicht so optimal, vor allem, nachdem es ja schon das private-WLAN gibt.

MichaelSasser, 2015-08-30:
Naja, ich halte davon auch nichts. Man stellt doch einen Freifunk Knoten auf, weil man irgendwo selbstlos ist und seine Ideale vertreten will. Wenn man damit werben will, kann man in der Community überlegen ob man Aufkleber drucken lässt und sich einen an die Eingangstür klebt. Es gibt tonnenweise vorlagen für die "Free WiFi" Aufkleber in den verschiedenen Communitys.

rubo77, 2015-08-30:
Ich finde die Idee super, das ist genau das, was wir brauchen, um Cafés zu begeistern, die sonst nicht mitmachen würden. 

Wenn es keine perforrmance probleme macht sollten wir das einbauen, aber im expert mode, damit das wirklich nur benutzt wird, wenn es unebedingt nötig ist (die Freifunk-SSID ist ja schliesslich für uns wichtiger)

jplitza, 2015-08-30:
Die Diskussion gab es schon in #303 und #141. Ergebnis: In gluon zentral wird das nicht geben, da die Kernentwickler das Feature nicht brauchen und nicht pflegen wollen. Jede Community kann es aber in einem eigenen [Community-Repo](http://gluon.readthedocs.org/en/v2014.4/user/site.html#packages) bauen und pflegen.

closed 2015-08-30 by NeoRaider