freifunk-gluon/gluon #475: 11s is broken in master due to recent mac80211 backports

tcatm, 2015-09-02:


tcatm, 2015-09-02:
```[  382.330000] ------------[ cut here ]------------
[  382.330000] WARNING: CPU: 0 PID: 6 at /home/nils/code/gluon/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/linux-ar71xx_generic/compat-wireless-2015-07-21/net/mac80211/ieee80211_i.h:963 mesh_matches_local+0xb0/0x2d4 [mac80211]()
[  382.350000] Modules linked in: ath9k ath9k_common iptable_nat ath9k_hw ath nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 mac80211 ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables ipv6 arc4 crypto_blkcipher gpio_button_hotplug
[  382.410000] CPU: 0 PID: 6 Comm: kworker/u2:0 Not tainted 3.18.20 #42
[  382.420000] Workqueue: phy0 ieee80211_ibss_leave [mac80211]
[  382.430000] Stack : 00000001 00000000 8783d500 8032b134 873d9360 873c1cb4 000003c3 873d4b84
	  87804790 87804780 873d4c50 800a3840 00000006 800962ac 8783af90 873d4b84
	  803347cc 87859bfc 87859bfc 800a3840 00000003 80093c9c 000003c3 00000000
	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
	  70687930 00000000 00000000 00000000 00000000 00000000 87298300 8783ea00
	  ...
[  382.460000] Call Trace:
[  382.470000] [<800710f0>] show_stack+0x48/0x70
[  382.470000] [<80080564>] warn_slowpath_common+0x84/0xb4
[  382.480000] [<8008061c>] warn_slowpath_null+0x18/0x24
[  382.480000] [<873c1cb4>] mesh_matches_local+0xb0/0x2d4 [mac80211]
[  382.490000] [<873c1e50>] mesh_matches_local+0x24c/0x2d4 [mac80211]
[  382.500000] 
[  382.500000] ---[ end trace 9351d618fb01c1ad ]---
[  382.500000] ------------[ cut here ]------------
[  382.510000] WARNING: CPU: 0 PID: 6 at /home/nils/code/gluon/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/linux-ar71xx_generic/compat-wireless-2015-07-21/net/mac80211/ieee80211_i.h:963 mesh_pathtbl_unregister+0x6d4/0xe44 [mac80211]()
[  382.530000] Modules linked in: ath9k ath9k_common iptable_nat ath9k_hw ath nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 mac80211 ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables ipv6 arc4 crypto_blkcipher gpio_button_hotplug
[  382.590000] CPU: 0 PID: 6 Comm: kworker/u2:0 Tainted: G        W      3.18.20 #42
[  382.600000] Workqueue: phy0 ieee80211_ibss_leave [mac80211]
[  382.610000] Stack : 00000001 00000000 8783d500 8032b134 873d9674 873c59f0 000003c3 873d4b84
	  00000001 87804780 873d4c50 800a3840 00000006 800962ac 8783af90 873d4b84
	  803347cc 87859b9c 87859b9c 800a3840 00000003 80093c9c 000003c3 00000000
	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
	  70687930 00000000 00000000 00000000 00000000 00000000 87298300 8783ea00
	  ...
[  382.640000] Call Trace:
[  382.650000] [<800710f0>] show_stack+0x48/0x70
[  382.650000] [<80080564>] warn_slowpath_common+0x84/0xb4
[  382.660000] [<8008061c>] warn_slowpath_null+0x18/0x24
[  382.660000] [<873c59f0>] mesh_pathtbl_unregister+0x6d4/0xe44 [mac80211]
[  382.670000] [<8738544c>] sta_info_move_state+0x2e8/0x304 [mac80211]
[  382.680000] [<873c5d6c>] mesh_pathtbl_unregister+0xa50/0xe44 [mac80211]
[  382.680000] 
[  382.690000] ---[ end trace 9351d618fb01c1ae ]---
[  382.690000] ------------[ cut here ]------------
[  382.700000] WARNING: CPU: 0 PID: 6 at /home/nils/code/gluon/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/linux-ar71xx_generic/compat-wireless-2015-07-21/net/mac80211/rate.c:49 rate_control_rate_init+0x5c/0xec [mac80211]()
[  382.720000] Modules linked in: ath9k ath9k_common iptable_nat ath9k_hw ath nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 mac80211 ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT nf_reject_ipv4 nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables ipv6 arc4 crypto_blkcipher gpio_button_hotplug
[  382.780000] CPU: 0 PID: 6 Comm: kworker/u2:0 Tainted: G        W      3.18.20 #42
[  382.780000] Workqueue: phy0 ieee80211_ibss_leave [mac80211]
[  382.790000] Stack : 00000001 00000000 8783d500 8032b134 873d4eb8 87395dd0 00000031 8790c238
	  00000001 00000000 873d4c50 800a3840 00000006 800962ac 8783af90 8790c238
	  803347cc 87859b6c 87859b6c 800a3840 00000003 80093c9c 00000031 00000000
	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
	  70687930 00000000 00000000 00000000 00000000 00000000 87298300 8783ea00
	  ...
[  382.830000] Call Trace:
[  382.830000] [<800710f0>] show_stack+0x48/0x70
[... cut ...]
	  872a4d90 00000001 0000096c 00000000 00000001 86d7a4c0 86c96370 873e4d18
	  00000080 873c1e64 87392590 87804790 87804780 800f961c 00000000 00000000
	  ...
[  383.070000] Call Trace:
[  383.070000] [<801a6514>] memcmp+0x10/0x30
[  383.070000] [<873c60b4>] mesh_pathtbl_unregister+0xd98/0xe44 [mac80211]
[  383.070000] 
[  383.070000] 
Code: 10660008  00001021  00831021 <90470000> 00a31021  90420000  00e21023  1040fff8  24630001 
[  383.300000] ---[ end trace 9351d618fb01c1b1 ]---
[  383.300000] Kernel panic - not syncing: Fatal exception in interrupt
```

tcatm, 2015-09-02:
Trunk
```
[  180.176360] ------------[ cut here ]------------
[  180.181319] WARNING: CPU: 0 PID: 1262 at /home/buildbot/slave/ar71xx/build/build_dir/target-mips_34kc_musl-1.1.10/linux-ar71xx_generic/compat-wireless-2015-07-21/net/mac80211/ieee80211_i.h:963 ieee80211_debugfs_key_sta_del+0xe4/0x6d0 [mac80211]()
[  180.203805] Modules linked in: ath9k ath9k_common pppoe ppp_async iptable_nat ath9k_hw ath pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 mac80211 ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_ftp nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbdev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables ipv6 arc4 crypto_blkcipher ohci_platform ohci_hcd ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[  180.278699] CPU: 0 PID: 1262 Comm: kworker/u2:2 Tainted: G        W       4.1.6 #1
[  180.286853] Workqueue: phy0 ieee80211_ibss_leave [mac80211]
[  180.292713] Stack : 86e0dd1c 87804400 872d5780 800a6878 80347cf0 803a4123 87a579d0 000004ee
	  803591c4 86e0db4c 86e0dd1c 800a4e88 872d5780 800a6878 00000003 8034d070
	  000003c3 86e0db4c 8034b584 80095070 872d5780 00000000 00000000 00000000
	  00000000 00000000 87292558 87bdb900 87bdb800 70687930 00000000 00000000
	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
	  ...
[  180.329805] Call Trace:
[  180.332311] [<80071c88>] show_stack+0x50/0x84
[  180.336744] [<80081968>] warn_slowpath_common+0xa0/0xd0
[  180.342231] [<80081a20>] warn_slowpath_null+0x18/0x24
[  180.347440] [<872c2450>] ieee80211_debugfs_key_sta_del+0xe4/0x6d0 [mac80211]
[  180.354885] [<800a6878>] vprintk_default+0x24/0x30
[  180.359950] [<800a4e88>] printk+0x2c/0x38
[  180.364030] [<80081970>] warn_slowpath_common+0xa8/0xd0
[  180.369542] [<80081a20>] warn_slowpath_null+0x18/0x24
[  180.374898] [<872c2bc4>] mesh_matches_local+0xf0/0x2d4 [mac80211]
[  180.381236] [<872c2d84>] mesh_matches_local+0x2b0/0x2d4 [mac80211]
[  180.387526] 
[  180.389040] ---[ end trace 3c214c742987c0e3 ]---
```

tcatm, 2015-09-02:
Loop when 11s mesh has no neighbour
```
[  391.319351] ibss: Creating new IBSS network, BSSID e6:56:6d:8e:22:4b
[  391.325966] ibss: Failed to join IBSS, no channel context
[  391.331982] ibss: Creating new IBSS network, BSSID 0a:0e:8f:dd:3d:73
[  391.338667] ibss: Failed to join IBSS, no channel context
[  391.344616] ibss: Creating new IBSS network, BSSID c2:1c:8a:6b:1a:0a
[  391.351148] ibss: Failed to join IBSS, no channel context
[  391.357153] ibss: Creating new IBSS network, BSSID 3e:2d:07:51:2e:8a
[  391.363688] ibss: Failed to join IBSS, no channel context
[  391.369545] ibss: Creating new IBSS network, BSSID 8a:2f:a2:1a:8d:d7
[  391.376143] ibss: Failed to join IBSS, no channel context
[  391.382212] ibss: Creating new IBSS network, BSSID be:54:76:10:13:2e
[  391.388847] ibss: Failed to join IBSS, no channel context
[  391.394875] ibss: Creating new IBSS network, BSSID f6:da:a1:2e:20:21
[  391.401414] ibss: Failed to join IBSS, no channel context
[  391.407697] ibss: Creating new IBSS network, BSSID 1a:89:4e:4b:af:e0
[  391.414232] ibss: Failed to join IBSS, no channel context
[  391.420245] ibss: Creating new IBSS network, BSSID 2e:71:bb:2f:b1:56
[  391.426855] ibss: Failed to join IBSS, no channel context
[  391.432784] ibss: Creating new IBSS network, BSSID 0a:ae:cb:d7:a9:b4
[  391.439342] ibss: Failed to join IBSS, no channel context
[  391.449061] ibss: Creating new IBSS network, BSSID a6:dd:d6:a1:71:87
[  391.455596] ibss: Failed to join IBSS, no channel context
[  391.461687] ibss: Creating new IBSS network, BSSID aa:60:a1:18:5e:8a
[  391.468276] ibss: Failed to join IBSS, no channel context
[  391.474309] ibss: Creating new IBSS network, BSSID b6:de:ce:8b:48:7f
[  391.480837] ibss: Failed to join IBSS, no channel context
[  391.486910] ibss: Creating new IBSS network, BSSID b2:a0:2b:d3:b0:75
[  391.493445] ibss: Failed to join IBSS, no channel context
[  391.499518] ibss: Creating new IBSS network, BSSID 6a:bb:b6:85:c0:8c
[  391.506120] ibss: Failed to join IBSS, no channel context
[  391.520624] ibss: Creating new IBSS network, BSSID ea:cd:bd:45:f1:bc
[  391.527281] ibss: Failed to join IBSS, no channel context
[  391.533196] ibss: Creating new IBSS network, BSSID 92:ab:f3:a5:70:15
[  391.539743] ibss: Failed to join IBSS, no channel context
[  391.545792] ibss: Creating new IBSS network, BSSID ba:fd:d4:d2:1b:ca
[  391.552334] ibss: Failed to join IBSS, no channel context
[  391.558290] ibss: Creating new IBSS network, BSSID 9e:53:2d:92:44:8e
[  391.564825] ibss: Failed to join IBSS, no channel context
[  391.570833] ibss: Creating new IBSS network, BSSID 76:1d:10:8b:62:a7
[  391.577426] ibss: Failed to join IBSS, no channel context
[  391.583367] ibss: Creating new IBSS network, BSSID 52:7d:50:6e:11:a7
[  391.589911] ibss: Failed to join IBSS, no channel context
[  391.595937] ibss: Creating new IBSS network, BSSID e2:26:5f:f9:2b:23
[  391.602473] ibss: Failed to join IBSS, no channel context
[  391.608495] ibss: Creating new IBSS network, BSSID 16:5c:35:bf:2b:fb
[  391.615038] ibss: Failed to join IBSS, no channel context
[  391.621344] ibss: Creating new IBSS network, BSSID 7a:0a:87:80:42:5e
[  391.628047] ibss: Failed to join IBSS, no channel context
[  391.634052] ibss: Creating new IBSS network, BSSID aa:74:4b:ce:4f:96
[  391.640590] ibss: Failed to join IBSS, no channel context
[  391.646824] ibss: Creating new IBSS network, BSSID d6:e6:70:00:ad:61
[  391.653354] ibss: Failed to join IBSS, no channel context
[  391.659540] ibss: Creating new IBSS network, BSSID fe:8e:03:95:8a:50
[  391.666198] ibss: Failed to join IBSS, no channel context
[  391.672246] ibss: Creating new IBSS network, BSSID fa:5c:06:37:09:63
[  391.678821] ibss: Failed to join IBSS, no channel context
[  391.684915] ibss: Creating new IBSS network, BSSID 4a:78:71:d5:3e:02
[  391.691444] ibss: Failed to join IBSS, no channel context
[  391.697439] ibss: Creating new IBSS network, BSSID 7e:0a:f5:65:05:ec
[  391.703980] ibss: Failed to join IBSS, no channel context
[  391.710017] ibss: Creating new IBSS network, BSSID 02:cb:30:09:4b:fa
[  391.716621] ibss: Failed to join IBSS, no channel context
[  391.722636] ibss: Creating new IBSS network, BSSID aa:75:be:3b:7d:8a
[  391.729255] ibss: Failed to join IBSS, no channel context
[  391.735546] ibss: Creating new IBSS network, BSSID fe:7c:13:2e:61:49
[  391.742104] ibss: Failed to join IBSS, no channel context
[  391.757001] ibss: Creating new IBSS network, BSSID a6:f7:ab:9d:7e:d0
[  391.763646] ibss: Failed to join IBSS, no channel context
[... 98 lines cut ...]
[  392.407498] ibss: Creating new IBSS network, BSSID 16:26:88:9c:91:d7
[  392.414040] ibss: Failed to join IBSS, no channel context
[  392.420058] ibss: Creating new IBSS network, BSSID c6:d4:95:d0:2e:24
[  392.426610] ibss: Failed to join IBSS, no channel context
[  392.432570] ibss: Creating new IBSS network, BSSID 66:e3:ca:48:62:7d
[  392.439150] ibss: Failed to join IBSS, no channel context
[  392.445123] ibss: Creating new IBSS network, BSSID 0e:53:f4:62:6b:53
[  392.451653] ibss: Failed to join IBSS, no channel context
[  392.457626] ibss: Creating new IBSS network, BSSID 02:e1:8d:53:3c:b1
[  392.464172] ibss: Failed to join IBSS, no channel context
[  392.470556] ibss: Creating new IBSS network, BSSID f2:3c:26:06:9f:c1
[  392.477235] ibss: Failed to join IBSS, no channel context
[  392.483338] ibss: Creating new IBSS network, BSSID ae:9a:f9:6a:d2:e2
[... cut ...]
[  392.516509] ibss: Failed to join IBSS, no channel context
[  392.522541] ibss: Creating new IBSS network, BSSID 1a:f9:e8:44:b9:23
[  392.529128] ibss: Failed to join IBSS, no channel context
[  392.535100] ibss: Creating new IBSS network, BSSID 0a:27:0b:1a:49:4d
[  392.541625] ibss: Failed to join IBSS, no channel context
[  392.547575] ibss: Creating new IBSS network, BSSID 6a:c6:2b:ee:76:09
[  392.554112] ibss: Failed to join IBSS, no channel context
[  392.560178] ibss: Creating new IBSS network, BSSID 4e:2d:70:56:da:b6

tcatm, 2015-09-03:
40b1a4590c6cd2e5d14c8555017970ad9c124440
```
[   20.379870] =============================================
[   20.385340] [ INFO: possible recursive locking detected ]
[   20.390820] 4.1.3 #2 Not tainted
[   20.394097] ---------------------------------------------
[   20.399572] ifconfig/919 is trying to acquire lock:
[   20.404515]  (&irq_desc_lock_class){-.-...}, at: [<800bb0f4>] __irq_get_desc_lock+0x88/0xb4
[   20.413033] 
[   20.413033] but task is already holding lock:
[   20.418955]  (&irq_desc_lock_class){-.-...}, at: [<800bb0f4>] __irq_get_desc_lock+0x88/0xb4
[   20.427457] 
[   20.427457] other info that might help us debug this:
[   20.434084]  Possible unsafe locking scenario:
[   20.434084] 
[   20.440093]        CPU0
[   20.442574]        ----
[   20.445056]   lock(&irq_desc_lock_class);
[   20.449130]   lock(&irq_desc_lock_class);
[   20.453206] 
[   20.453206]  *** DEADLOCK ***
[   20.453206] 
[   20.459216]  May be due to missing lock nesting notation
[   20.459216] 
[   20.466101] 3 locks held by ifconfig/919:
[   20.470163]  #0:  (rtnl_mutex){+.+.+.}, at: [<802f26a0>] devinet_ioctl+0x178/0x78c
[   20.477898]  #1:  (&sc->mutex){+.+...}, at: [<869e37cc>] ath_cancel_work+0x8c/0x3cc [ath9k]
[   20.486434]  #2:  (&irq_desc_lock_class){-.-...}, at: [<800bb0f4>] __irq_get_desc_lock+0x88/0xb4
[   20.495393] 
[   20.495393] stack backtrace:
[   20.499828] CPU: 0 PID: 919 Comm: ifconfig Not tainted 4.1.3 #2
[   20.505830] Stack : 00000000 00000000 00000000 00000000 803adfcc 80410703 87964328 00000397
	  80673588 87964618 80ae1050 80420000 00000002 800ba100 00000002 00000002
	  00000000 00000000 803b3f34 8650fa5c 80e203d0 800b8448 00000002 00000000
	  80ae1050 00000000 00000000 00000000 00000000 00000000 00000000 00000000
	  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
	  ...
[   20.542165] Call Trace:
[   20.544659] [<80074cfc>] show_stack+0x48/0x70
[   20.549089] [<800b21fc>] __lock_acquire+0x7c8/0x1a90
[   20.554134] [<800b3c74>] lock_acquire+0x60/0x88
[   20.558740] [<8006b39c>] _raw_spin_lock_irqsave+0x54/0x78
[   20.564226] [<800bb0f4>] __irq_get_desc_lock+0x88/0xb4
[   20.569446] [<800bc26c>] __disable_irq_nosync+0x24/0x68
[   20.574751] [<800bc2cc>] disable_irq+0x14/0x38
[   20.579267] [<800bdd70>] irq_disable+0x34/0x54
[   20.583781] [<800bc280>] __disable_irq_nosync+0x38/0x68
[   20.589087] [<800bc2cc>] disable_irq+0x14/0x38
[   20.593618] [<869e4748>] ath9k_calculate_summary_state+0x4e8/0x650 [ath9k]
[   20.600627] [<869eccac>] ath_chanctx_set_channel+0x178/0x78c [ath9k]
[   20.607103] [<869e3934>] ath_cancel_work+0x1f4/0x3cc [ath9k]

CodeFetch, 2015-10-26:
@NeoRaider Can you apply the patches for WR941ND v6 now?

NeoRaider, 2015-10-26:
@CodeFetch, yes, as soon as the mac80211 backport has been merged into master.

PR #531 (2015-11-02): mac80211: backport from trunk r47249 + additional patches
Fixes #475.
closed 2015-11-02 by NeoRaider