freifunk-gluon/gluon #476: option to regularly restart a node

rubo77, 2015-09-03:
it would be nice to have an option to define regularly reboots every X days.

NeoRaider, 2015-09-03:
This would just hide sporadic bugs, making debugging even harder.

closed 2015-09-03 by NeoRaider

bittorf, 2015-09-04:
in our firmware we have via _default_ a hard reboot after 1.5 days of uptime. this help to recover from very bad situations where nothing works anymore. this can be overriden via uci. we stick on this feature and it is easy to implement:

read UP REST </proc/uptime || UP=  # e.g. procfs broken
[ ${UP%.*} -gt 169200 -o -z "$UP" ] &&
  bool_true 'system.@system[0].avoid_autoreboot' || reboot
}

FreifunkUFO, 2017-07-12:
the ticket was closed, so please tell us the name of the gluon-package or further documentation to make it with gluon.

@bittorf yes, also in our (non-gluon) firmware we have an option for daily reboot, and i wont miss it
the "reboot" command isnt that nice, i prefer "reboot -f" because sometimes batman-adv kernelmodul was preventing the reboot (while rebooting)