freifunk-gluon/gluon #477: WiFi offline times

rubo77, 2015-09-03:
some people would like to have their WiFi off during night time

RubenKelevra, 2015-09-03:
They should get some tinfoil hats for free. :)

benchlor, 2015-09-09:
this can already be done by a cronjob, like:
0 22 * * *  uci set wireless.client_radio0.disabled=1;uci set wireless.client_radio1.disabled=1;wifi
0  6 * * *   uci set wireless.client_radio0.disabled=0;uci set wireless.client_radio1.disabled=0;wifi
(posted here https://forum.freifunk.net/t/herumlungernde-jugendgruppen/7723/4 )
this is an example for a dualband router, singleband router only need the client_radio0 command.

closed 2015-09-09 by tcatm

smoe, 2016-03-22:
I like that feature. Albeit I also liked the distribution of tinfoil hats. Not that the core Gluon folks should invest much of their time into it, but the support of such a time-constraint module for the wizard pages seems appropriate to me.

belzebub40k, 2016-03-22:
+1 for this request. Currently I also do it via cronjob but this has two big disadvantages.
- Does not survive updates
- It's not enduser friendly

Gammelknup, 2016-03-22:
we have to use the cronjob-wireless.client_radio-solution too. 
Usecase: To avoid noise at night times(0:00-6:00) in gyms witch are used as refugeecamp.

fpletz, 2016-03-24:
In what universe were social problems ever resolved by technical means?

And shutting down internet for refugees? Sorry for the language, but what the fuck is wrong with you people? The internet doesn't make noise. People do. Educate them individually instead of fucking with all of them to play the pawns in your power games. :-1:

I don't see any technically useful, health-protective or morally justifiable use case to support this feature.

corny, 2016-03-24:
I have also been asked for this feature. Even if it existed I won't enable it. My advice for the camp:
https://www.obi.de/decom/product/OBI_Wochenzeitschaltuhr_digital_mit_98_Schaltmoeglichkeiten_Weiss/9680471

belzebub40k, 2016-03-24:
We also have problems at some nodes that people sit around after 10pm and make a lot of noise and the owner was forced to shut them down to not risk a termination of his lease. Why should I add an additional device if we can implement this a software feature?

Maybe it would also be a cool idea to enhance this feature with dynamic bandwidth limits. As I don't care if my line is used to 100% when I'm not at home but I do when I'm.

Gammelknup, 2016-03-24:
@fpletz you are absolutely right - in a perfect world. 
In camps where hundreds of refugees live in one gym, noise is a big problem. And please consider the persons in those temporary camps, live there only for a few weeks, before they are moved to more appropriate locations. If you want to teach them you have to do it every day, because every day about 10% of the people change. At the night times there is only one security guy and to keep social harmony is not an easy job. 
My first reaction, when the DRK asks us to turn the client wifi off in the times between 0:00 and 6:00 were the same as yours. We have spoken with the refugees and most of them are more happy with this solution, because at night times social harmony is more important than wifi.
From a freifunk perspective i can live with this solution as well, because mesh wifi is always turned on and so mesh nodes around can do free transit of data.

belzebub40k, 2016-03-24:
@Gammelknup That is exactly the same experience I made. I think the last part is the most important part. If you have a node with an uplink you can leave the mesh running and only deactivate the client WiFi which would also be killed in case of a timer switch.