freifunk-gluon/gluon #479: Improve wifi-bandwidth with coalesce-rules?

RubenKelevra, 2015-09-03:
I think it's a nice idea to combine the broadcast, to enhance battery-life on the mobile devices and reduce overhead for broadcasts :)

https://wireless.wiki.kernel.org/en/users/Documentation/iw#creating_packet_coalesce_rules

jplitza, 2015-09-03:
* This feature is currently disabled by `openwrt/package/network/utils/iw/patches/200-reduce_size.patch`. Removing that whole patch grows the `iw` binary by ~11KB. (And *man* is `iw phy` much longer afterwards...)
* My WDR4300 simply reports "command failed: Operation not supported (-122)" for `iw phy phy0 coalesce show`. Do you have any model/chipset that actually supports coalescing? Otherwise we can ignore that feature until hardware support is there.

RubenKelevra, 2015-09-03:
Well thanks for the infos, thought that it's a feature which is implemented in the kernel not in the drivers.

I think the ticked can be closed since some support appear.

closed 2015-09-03 by jplitza