freifunk-gluon/gluon #484: Feature-Req: email alert wanted

rubo77, 2015-09-09:
An extra checkbox field that is transmitted via alfred, to announce, that you wish to be contacted via email, whenever your server is not reacheable through the internet.

This could be a checkbox, or a field that defines after how many hours offline you want to be alerted

do9xe, 2015-09-13:
This is a request for a implementation of a monitoring-system. If you wish to monitor your nodes you need to set up a system by yourself, so this issue is going to be closed.
There is no need to set up a monitoring-system, which is implemented in gluon by default.

jplitza, 2015-09-13:
I developed [something like this](https://github.com/FreifunkBremen/nodewatcher/) a while ago (supporting mail, twitter, xmpp, IRC and pluggable other notification methods), but didn't maintain it in a long time. However, I plan on reviving it.

rubo77, 2015-09-13:
In gluon there only would be to have a hook to announce that you want to take part in the monitoring system, (which has to be external)

NeoRaider, 2015-09-14:
The monitoring system does currently not exist, and there is no need to enable such a thing in Gluon itself. When a monitoring system has been developed, we could think about adding such a hook, but for now, this doesn't make any sense.

closed 2015-09-14 by NeoRaider

rubo77, 2015-09-14:
please add a flag `rejected` or something like `enhancement` if you plan to implement this some day

jplitza, 2015-09-14:
If I ever get around to update my monitoring system, I'll probably implement it in [our community repo](https://github.com/FreifunkBremen/ffhb-packages/). But as this is highly community-dependent and almost trivial to implement, it will not be implemented in the core of Gluon.