freifunk-gluon/gluon #486: Feature Request: quota for total data up-/down-load per month

Echsecutor, 2015-09-09:
I was surprised to learn that contracts limiting the total up-/down- stream per month still exist for landlines. Anyway, this being the case, someone with such a contract obviously wants to limit the total up/down stream rather than the bandwidth used by gluon.

Hence it would be nice to have an option like: use full bandwidth for up to X GB of total up/down stream, after that is used reduce bandwidth (limit to Y) + reset back to full at the end of the month.

tcatm, 2015-09-09:
Someone with these kind of contracts could easily get a contract that works better for their use case. Such a feature is out of scope for Gluon. We're building a wireless mesh firmware.

jplitza, 2015-09-09:
Actually, I don't see why a person with limited high speed bandwidth should connect the node to the internet *at all*. Every byte that the node uses is subtracted from the high speed quota, and if the volume Y that you propose is used up at high or low bandwidth doesn't matter at all. So limiting to a low bandwidth after reaching Y doesn't change anything.

So I totally agree with @tcatm: People with low bandwidth (i.e. below at least 8 MBit/s) or people with volume contracts shouldn't connect a Freifunk node to the internet. To the contrary, these people probably are candidates who would benefit from using Freifunk offered by others.

Echsecutor, 2015-09-09:
Given a perfect world in which a wireless mesh exists, you are absolutely right. Unfortunately the situation here is that I am trying to convince individuals to set up isolated nodes as germs for a yet to grow mesh. In such a situation the choice is either being your own uplink or postponing the idea altogether until your neighbours will have been kind enough to start the mesh.

Accepting your arguments, the existing limit bandwidth option could be questioned along the same lines, right?

Anyway, thanks for your feedback, I will advise the person under consideration to not become a part of "Freifunk", yet.

closed 2015-09-09 by Echsecutor