freifunk-gluon/gluon #490: SIS5513 driver support

christian1980nrw, 2015-09-11:
Hello, many of us are ordering the futro thin client now to use it as offloader . It has a internal compact flash and this is needing the sis5513 IDE driver. Could you integrate it into the kernel of the usb gluon Image by default? Thanks. 
https://forum.freifunk.net/t/80mbit-download-thin-client-siemens-futro-s550-als-ofloader/7711/

christian1980nrw, 2015-09-11:
http://kernel.xc.net/html/linux-2.6.29/x86/BLK_DEV_SIS5513

christian1980nrw, 2015-09-11:
I think its included in the driver named pata_sis

NeoRaider, 2015-09-14:
This should be addressed in the OpenWrt upstream first.

mruettgers, 2015-09-17:
The Futro S550 has an ATI SB600 IDE-Controller built in and the missing kernel module is pata_atiixp.

To build the driver you have to add it to the openwrt target configuration:
echo "CONFIG_PATA_ATIIXP=y" >> openwrt/target/linux/x86/generic/config-x.yy

But this should indeed be addressed in OpenWrt upstream, as I have not found a way to pass kernel config settings from Gluon to OpenWrt.

A-Kasper, 2015-09-17:
I have also ordered 75 Futro S550 today. They come with 1 GB CF. It would be very nice to be able to use the CF instead of USB-sticks. 
If I'm right the way to pass the kernel config settings in Gluon would be to build a patch with this change and make an pr to the patches/openwrt directory. 
Is it really necessary to submit it to OpenWRT in general?
Maybe we could just add another build target via a gluon patch to build special Futro S550 images. I don't think, that OpenWRT would accept PRs for special x86 hardware setups. I don't think that this enhancement is something that should be addressed in the OpenWRT upstream.

NeoRaider, 2015-09-17:
I see no problem with OpenWrt accepting such a patch if it is formally correct (has a nice commit message etc.), the x86 target is not as minimal as the ar71xx targets and OpenWrt is trying to suppport all possible hardware.

For Gluon, we have the rule not to accept hardware support patches that have not been reviewed (and hopefully accepted) upstream, so we can drop them when we switch to the next OpenWrt release. Of the 24 patches in `patches/openwrt`, only 9 are Gluon-specific, the rest has been accepted upstream or is at least in the review loop. (I guess we could even reduce that number further with some work...)

A-Kasper, 2015-09-18:
Hi Matthias,

thanks for explanation. I was wrong. I thought that OpenWRT should make
a special Futro image instead of adding hardware to the x86 image.

Because of high number of devices I would like to prepare a temporary
Futro "Test"-Fork to make all these devices usable as fast as possible.
Just for the actual stable gluon and as short time solution to give
something to the owners they would be able to use. Solving the problem
seems to be nothing more than a one-liner. Until the next build, the
patch should have managed the OpenWRT review process. And the as
unstable-marked testfork could be removed....

Would you agree with this proceeding?

NeoRaider, 2015-09-18:
As long as the patch is also submitted upstream, I'm not opposed to adding it to Gluon as well, this can be done in the Gluon master. I just don't want patches to sit around in the Gluon repo forever (like the hundreds of kernel patches in OpenWrt ...)

Have you checked that the device is fully functional after CONFIG_PATA_ATIIXP is added? I could prepare and submit the patch myself, but I'd prefer if someone who actually has such a device to test it could do it.

kaechele, 2015-09-18:
I could prepare and submit the patch this weekend if that is quick enough.

A-Kasper, 2015-09-18:
My devices didn't reach me yet. But others in this "internet" told they
fixed it this way. I will have to validate that it boots through CF. If
that works I would say that it works. Maybe someone else here is already
able to validate it...

I would prefere a fork with a backport, to stay in the same stable branch
used in my community, but I would be able to backport this on my own.

mruettgers, 2015-09-18:
I have this device here, that's how I figured out the missing kernel driver.
Booting from the CF-Card works flawlessly here with CONFIG_PATA_ATIIXP.

kaechele, 2015-11-04:
Changes are now upstream:
https://github.com/openwrt-mirror/openwrt/commit/aa9b7c1be36a4a9b3a22547ff6f442b87b9e44d4
https://github.com/openwrt-mirror/openwrt/commit/ae9f4943679439f471d7b8044ce03e944d5b5766
https://github.com/openwrt-mirror/openwrt/commit/1052e5914d2680398f3d10bbc9c0a88c90f2fe5b

NeoRaider, 2015-11-04:
@kaechele, thanks... but does compling radeon into the kernel for x86-generic actually make sense? Wouldn't it be better to make it an optional module?

kaechele, 2015-11-23:
It does make sense because that enables modesetting at an early stage in the boot process. Of course, if you don't want or like that it doesn't make sense.
On x86 we don't have space constraints so upstream accepted the change under the stance that a broad support of different hardware platforms is desirable.
For a backport from upstream for the sake of getting the Futros to boot from their internal CF card the first two patches would suffice.

NeoRaider, 2015-11-23:
@kaechele If we consider x86 "without space constraints", there shouldn't be a distinct x86-kvm_guest target either... well, I don't really care.

Can you prepare a pull requests with the backports?

PR #646 (2016-02-08): backport: Futro support in x86-generic

Fixes #490
closed 2016-02-08 by NeoRaider