freifunk-gluon/gluon #499: Add Firewall options

rubo77, 2015-09-15:
The idea of Freifunk is, to provide a complete uncensored Network. 

So at the moment, if you put a Freifunk router at some private place, that has no other internet connection, the Freifunk router works at the same level as a **cable modem** which lets the clients  unfiltered and directly into the Freifunk-net, which, if large, has the same menace as the Internet. Who would do such a madness at home? And Layer2 a new user does not expect ever!

Strictly speaking, you would have to put another router behind the Freifunk router to get a similar security, as I know it from the home LAN.

Could we add some settings in the config mode to enable some basic Firewall?

Or maybe start an optional second Wifi which is protected with another SSID?

jplitza, 2015-09-15:
You could, we (or at least I) won't. You stated the reason yourself: Freifunk is a completely uncensored network.

In fact, with public IPv6, every device is reachable from the internet as well, not only from the Freifunk network. And this is no different in university networks where you even get public IPv4 addresses on your device.

And there is no need for another plastic box that acts as a firewall: Every operating system either is completely locked down network-wise or ships with a firewall itself, which should be more than enough to secure your device. Furthermore, Freifunk is not only about providing network access, it is also about technical education, meaning that we do not want to have the "connect and forget" attitude, but we want users to be aware of the risks lurking in networks of all kinds and how to cope with them.

NeoRaider, 2015-10-11:
Having a setting in the config mode doesn't make any sense. In most cases, the users of a Freifunk node aren't the same people as the node's owners.

The firewall needs to be on the client device to have any effect (as the Freifunk network is completely open and thus not significatly safer than the internet itself). Windows has such a firewall by default, and other OSs's default configurations are more or less secure as well.

closed 2015-10-11 by NeoRaider