freifunk-gluon/gluon #501: Notice to use decimal coordinates

andi98, 2015-09-21:
Someone who set up his*her own Freifunk-Node contacted us with the please to add a notice in config mode which says that decimal coordinates have to be used.
This shouldn't be too difficult, should it?

jplitza, 2015-09-22:
This would go along the lines of what I said in #453, namely that not only the altitude field needs better specification. Thus, +1

NeoRaider, 2015-10-11:
There's an example below the coordinate fields... should that be enough (and even more useful than a notice?).

Adorfer, 2015-10-29:
My preffered solution would be a coordnate parser taking even lat/lon from the same input field, and accepting different formats like
DDD.DDDDD DDD.DDDDD
DDD,DDDDD DDD,DDDDD
DDD MM.MMM DDD MM.MMM
DDD MM SS DDD MM SS 
There is some fuzzuy magic behind this shown in https://github.com/dbarbalato/magellan
(but perhaps this a bit overdone for the purpose.)

mweinelt, 2020-03-02:
This can be configured in your respective local site repositories i18n files. The language strings are:

- msgid "gluon-config-mode:altitude-label"
- msgid "gluon-config-mode:altitude-help"

closed 2020-03-02 by mweinelt