freifunk-gluon/gluon #502: VMware x86-generic/x86-64 - IPv6 does not work

Datafreak, 2015-09-23:
Hi,

I hope someone can help me. I build the x86 Images from git master but the IPv6 does not work.
It looks like that the kernel modul can not be loaded. I use VMware vSphere Hypervisor 6 with a USB network card (asix-ax88179). I build this packages for it: kmod-libphy kmod-mii kmod-usb-net kmod-usb-net-asix-ax88179 kmod-usb-core kmod-usb3 libusb-1.0 zlib usbutils

In the configuration mode everything look ok. But after I have configured the node and rebooted I have no IPv6:
[image]
[image]
[image]
[image]

Here the x86-64 image: https://www.dropbox.com/s/izu88dplkrsjs4f/gluon-ffka-master-x86-64-vmware.vmdk?dl=0

Can someone look on thi issue?

Thanks,
Datafreak

NeoRaider, 2015-09-24:
The image you provided works for me in VMware Workstation.

Datafreak, 2015-09-24:
Hi Matthias,

thank you for you feedback. I have now deleted my VMware vSphere Hypervisor 6 and installed VMware vSphere Hypervisor 5.5. And there it works. So it looks like, the Image is somehow incompatible to the newest version of vSphere.

So I think we can close this issue here.

Best Regards,
Datafreak

closed 2015-09-24 by Datafreak