freifunk-gluon/gluon #506: Firewall rule blocking RAs

nazco, 2015-09-27:
There is a IPv6 firewall rule in the gluon 2015.1.2 release blocking router advertisements from the gateway server.

To reproduce:
 - flash a router with gluon 2015.1.2
 - setup a freifunk gateway and announce RAs from it
 - no RAs recived on the gluon node nor on a client

If I disable the IPv6 firewall rules on the gluon node the RAs are recived by the node and the client and IPv6 works. But I couldn't find the rule which is responsible for this at the moment.

nazco, 2015-09-27:
Very strange, now everything works. It looks like it is no firewall rule causing this problem. I double checked it with a newly flashed node. But I can say for sure that the problem was existing for two whole days till now. I'm very confuse because I didn't change the firmware nor the gateway. The gateway is configured by an ansible playbook only so I know that I haven't changed anything manually. And also another person on the IRC is confirming this behavior on a newly created network with only one gateway and one node. I'll try this in the next days again to check if this happens again if I setup a Network with on Gateway and one node.

NeoRaider, 2015-10-11:
I doubt that the issue is the firewall. I'll try to reproduce the issue here.

NeoRaider, 2015-10-12:
I can't reproduce this issue here.

nazco, 2015-10-12:
If I find some spare time I'll test it again.

NeoRaider, 2015-10-26:
Closing this, please reopen if you can reproduce this.

closed 2015-10-26 by NeoRaider