freifunk-gluon/gluon #507: Bindestrich  in Peer-Name wift Fehler

MPW1412, 2015-09-28:
Hallo,

mir ist gerade aufgefallen, dass es einen Validierungsfehler gibt, wenn man im Namen eines peers einen Bindestrich benutzt.

Ich kann mir nicht erklären warum das ein Problem sein sollte, zumal man das Zeichen auch in URLs benutzen darf. 

Beispiel: https://github.com/FreiFunkMuenster/site-ffms/blob/master/site.conf

Verursacht:

```/var/lib/jenkins/jobs/site-ffms/workspace/gluon/build/ar71xx-generic/openwrt/staging_dir/host/bin/lua: ...e/gluon-core/files/usr/lib/lua/gluon/site_config.lua:9: [string "site.conf"]:78: '}' expected (to close '{' at line 50) near '='
stack traceback:
	[C]: in function 'assert'
	...e/gluon-core/files/usr/lib/lua/gluon/site_config.lua:9: in main chunk
	[C]: ?
Your site configuration did not pass validation.
/var/lib/jenkins/jobs/site-ffms/workspace/gluon/Makefile:289: recipe for target 'prepare' failed
make[1]: *** [prepare] Error 1
make[1]: Leaving directory '/var/lib/jenkins/jobs/site-ffms/workspace/gluon/build/ar71xx-generic/openwrt'
Makefile:69: recipe for target 'all' failed
make: *** [all] Error 2```

jplitza, 2015-09-28:
Die `site.conf` ist eine Lua-Datei und muss daher den Lua-Standards genügen. Mit
```patch
-					sn-fanlin-1 = {
+					["sn-fanlin-1"] = {
```
funktionieren auch Namen mit Bindestrichen, so direkt werden sie aber als Minus aufgefasst und es wird versucht eine Subtraktion von Unbekannten durchzuführen. Siehe dazu auch ["Lexical Convention"](http://www.lua.org/manual/5.1/manual.html#2.1) in Kombination mit ["Table Constructors"](http://www.lua.org/manual/5.1/manual.html#2.5.7) in der Lua-Dokumentation.

closed 2015-09-28 by jplitza

MPW1412, 2015-09-29:
Danke für die Info!

MPW1412, 2015-10-10:
Tja, die Syntax funktioniert übrigens nicht! Die Peers werden bei einem sysupgrade nicht angelegt.

Unterstriche funktionieren, die Syntax ["..."] nicht.

NeoRaider, 2015-10-11:
Das ist dann eine Einschränkung von uci.

MPW1412, 2015-10-11:
Ja, vermutlich. Daher hatte ich im #Gluon-IRC vorgeschlagen, beim Bauen der Images eine Abfrage zu machen, ob Bindestriche in den Peernamen sind.

NeoRaider, 2015-10-11:
Ich habe einen entsprechenden Check hinzugefügt.

MPW1412, 2015-10-11:
Danke.