freifunk-gluon/gluon #509: persistent flag für peers in site.conf

A-Kasper, 2015-10-05:
Das Problem:
Wenn die Firmware in eine andere Community gewechselt wird, werden die Peers nur dann gelöscht oder überschrieben, wenn in der site.conf ein Eintrag mit gleichem Namen ist. Hierdurch entstehen immer wieder ungewollte Brücken zwischen Netzwerken. 

Lösungsvorschlag:
Grundsätzlich werden alle in der site.conf hinterlegten Peers gelöscht und die in der neuen site.conf hinterlegten Peers eingetragen. Damit es auch weiterhin möglich ist zusätzliche Tunnel einzustellen, die nicht durch autoupdates gelöscht werden wird eine option "persistent=1" eingeführt. Mit dieser kann man Tunnel markieren die im updatefall beibehalten werden sollen.

tcatm, 2015-10-05:
Grundsätzlich muss man Firmwareupdates bei einem solchen Communitywechsel immer ohne Übernahme der Einstellungen vornehmen. Die Upgradeskripte können mit diesem Fall ansonsten nicht sinnvoll umgehen.

Das fastd Upgradeskript ersetzt alle in der `site.conf` definierten Gruppen indem es alle darin konfigurierten (d.h. nicht nur jene, die in der `site.conf` stehen) entfernt und anschließend entsprechend der `site.conf` neue Einträge anlegt. Gruppen, die nicht in `site.conf` stehen, werden nicht angefasst und bleiben erhalten.

Das "persistent=1" Feature lässt sich damit umsetzen.

closed 2015-10-05 by tcatm

A-Kasper, 2015-10-05:
Das ist die Theorie. Die Praxis zeigt dass immer wieder router bridgen.
Allein in den letzten 7 Tagen hatten wir 2 Fälle.

Wenn das fastd update Script hier probleme macht, müsste es doch möglich
sein ein eigenes zu verwenden. Es wird ja letztlich nur eine fastd config
aus der site.conf generiert.

Andere Lösungsideen? Ein Löschen ALLER Einstellungen ist auch nicht der
Königsweg, da wir zunehmend domainsplits haben. bei diesen muss man zzt.
genau wissen wie die peers der alten side.conf hießen.  Es muss doch
möglich sein die fastd conf standardmässig komplett zu ersetzen....

NeoRaider, 2015-10-05:
Die Upgrade-Scripte ersetzen jede Peer-Gruppe, die in der site.conf steht, vollständig. Man muss nicht wissen, wie die alten Peers heißen, nur wie die Gruppe heißt.

A-Kasper, 2015-10-05:
Ja das weiß man aber nicht :-)
Evtl. wäre eine Lösung den höchsten Groupname nicht einstellen zu können.
und alles andere drunter anzusiedeln. dann wäre sie bei allen identisch.