freifunk-gluon/gluon #511: x86-generic - doesn't boot on HP Compaq Pro 6000 Microtower

MPW1412, 2015-10-08:
Hello,

Gluon x86-generic doesn't boot from USB drive on the HP Compaq Pro 6000 Microtower. The board seems to be a proprietary system from HP. There is no model id on it.

Workaround: Syslinux works, copy the kernel from the boot partition, install syslinux, set the root device to /dev/sda2 and copy back the kernel. Then the system boots and works correctly.

Google doesn't have much information about this board and grub, only some forum entries, that it just doesn't work.

Any ideas what I could to make it work with grub?

Thanks for reading
Matthias

mweinelt, 2015-11-08:
More information on the target machine would be necessary to make any qualified statement:
`lspci`
`lsusb`
`dmidecode`

NeoRaider, 2017-02-23:
No reply, closing.

snoerenberg, 2017-04-20:
Hi, I have the same problem but no clue how to replace grub with syslinux. Can you help me there? Damn Small Linux with syslinux boots fine on an CV860A.

Thanks

A-Kasper, 2017-04-20:
You can read
http://www.syslinux.org/wiki/index.php?title=HowTos#How_to_Create_a_Bootable_USB:_Windows_XP_and_Vista
to build a syslinux usb drive. After booting it you should be able to
provide the requested information.

snoerenberg, 2017-04-21:
lspci:
'''
00:00.0 Host bridge: VIA Technologies, Inc. VT8601 [Apollo ProMedia] (rev 05)
00:01.0 PCI bridge: VIA Technologies, Inc. VT8601 [Apollo ProMedia AGP]
00:07.0 ISA bridge: VIA Technologies, Inc. VT82C686 [Apollo Super South] (rev 40)
00:07.1 IDE interface: VIA Technologies, Inc. VT82C586A/B/VT82C686/A/B/VT823x/A/C PIPC Bus Master IDE (rev 06)
00:07.2 USB Controller: VIA Technologies, Inc. VT82xxxxx UHCI USB 1.1 Controller (rev 1a)
00:07.3 USB Controller: VIA Technologies. Inc. VT82xxxxx UHCI USB 1.1 Controller (rev 1a)
00:07.4 Bridge: VIA Technologies, Inc. VT82C686 [Apollo Super ACPI] (rev 40)
00:08.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL-8139/8139C/8139C+ (rev 10)
00:09.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL-8139/8139C/8139C+ (rev 10)
00:ob.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL-8139/8139C/8139C+ (rev 10)
01:00.0 VGQ compatible controller: Trident Microsgstems CgberBlade/i1 (rev 6a)
'''
lsusb
'''
Bus 002 Device 001: ID 0000:0000
Bus 002 Device 001: ID 046d:c52b Logitech, Inc.
Bus 001 Device 001: ID 0000:0000
Bus 001 Device 002: ID 0781:5575 SanDisk Corp.
Bus 001 Device 003: ID 23a9:ef18
'''

I have really no experience how to get syslinux.cfg configured to boot correctly a actual gluon image. All tries ended in a kernel-panic during boot.

A-Kasper, 2017-04-21:
I would try to load via686a and pata_via in site.mk
(or check if they are loaded..)

snoerenberg, 2017-04-26:
´´´
 * opkg_install_cmd: Cannot install package via686a.
make[4]: *** [package_install] Error 255
´´´

Adorfer, 2017-04-26:
Genau, wie A-Kasper schreibt, nicht via opkg.

A-Kasper, 2017-04-27:
Just to explain: you choose to compile a driver in the kernel. That's why
you can't install it as package. just add the line in site.mk

you can add it just for x86 (you don't want the driver on other devices to
reduce the image size)

ifeq ($(GLUON_TARGET),$(filter $(GLUON_TARGET),x86-generic x86-kvm_guest
 x86-xen_domu x86-64))
# support the usb stack on x86 devices
# and add a few common USB NICs
GLUON_SITE_PACKAGES += \
kmod-usb-core \
kmod-usb2 \
kmod-usb-hid \
kmod-usb-net \
kmod-usb-net-asix \
kmod-usb-net-dm9601-ether\
        kmod-carl9170\
kmod-r8169 \
kmod-rtl8192cu\
kmod-usb-net-rtl8150\
kmod-usb-net-rtl8152\
kmod-ath9k-htc\
        kmod-via686a
endif

(this is the via686a extended snipplet of our site.mk  ...untested)

NeoRaider, 2017-04-27:
kmods are dynamic modules, they are loaded *after* the kernel has booted. You can't affect the kernel itself from site.mk. Therefore, these modules won't help with the boot at all.

The question is: does the kernel even start to run, or not? If it does, it will display an error message, or at least a last message before hanging. In that case, that may give us a hint to which kernel options to set. In that case, replacing the bootloader should not affect this at all. If on the other hand using another bootloader fixes this issue, it is most likely not a kernel issue.

A-Kasper, 2017-04-27:
sorry.. matthias ist right. when we added the drivers for futros in
openwrt/target/linux/x86/generic/config-default, the kmods where just for
additional drivers..... I've no clue, if there is a similar way to add
support for the needed hardware...

MPW1412, 2017-04-28:
I think, this is not a Gluon issue :). I suggest to close this again.

snoerenberg, 2017-04-28:
Hi,
I added it via site.mk but NeoRaider already explained why it still will not work.

I can get it to work (really slow) when I disable CPU Cache (L1).
@MPW1412 how have you replaced the Grub2 with Syslinux?

This is what I want to know :)

MPW1412, 2017-04-28:
I had, the installation is out of service now. But it worked with syslinux. So this is a bug in grub probably.

snoerenberg, 2017-04-28:
How could you exchange the grub with syslinux? Is there anything I can follow as a step by step tutorial?

rotanid, 2018-02-13:
closing, can still be used for discussion

closed 2018-02-13 by rotanid