freifunk-gluon/gluon #515: Support fpr TP-Link  TL-WR841ND v10

mmalte, 2015-10-10:
I've seen the first v10 Devices popping up in the wild.

Is anybody already working on the Support for this Device?

The Openwrt wiki tells us that it is quite similar to the v9 Device:

http://wiki.openwrt.org/toh/tp-link/tl-wr841nd#hardware_v10_is_identical_with_v9_except_the_cpu_at_650_mhz

NeoRaider, 2015-10-11:
I've added support for the v10. As it is untested, it has to be compiled with BROKEN=1 for now.

Sunz3r, 2015-10-15:
Thanks! That works for me. :)

Procinger, 2015-10-19:
Hi,

tested the latest https://github.com/freifunk-gluon/gluon/commit/563f9afcc0b38766b6f1a2b776a44aa0f6e80908 version.

Flashing etc. is working fine, but Wifi signal is poor.

Nexus 5 & Galay Nexus with "Wifi Analyzer App"  installed at 2,5 meters I receive between -80 - -85 dBm

heini66, 2015-10-19:
> tested the latest 563f9af

moin

identisches ist in hh auf der liste zu lesen.
zitat:

Ich wollte nur mal einen Knoten aufstellen – jetzt sitze ich hier mit 841
V10 L

Die firmware 0.8~exp20151017 / gluon-v2015.1-208-g563f9af von Björn läuft.

Sendeleistung ist schlecht – auch mit 8db-Antennen von tp-link.

Meshverbindungen reißen immer wieder ab.

zitatende

moin aus der lüneburger heide

Marssl78, 2015-10-19:
Hi!
I have two TP-Link TL-WR841N v10 here, which I wanted to use as devices for my local Freifunk community, but they just offer Firmwares until V9. Searching the net for some support I found these two threads, where users modified the V9-Firmware in order to be able to flash it on V10-devices. Everything seems to work fine unless the two external Antennas are not working, just the internal one seems to work.

https://forum.openwrt.org/viewtopic.php?id=60130

https://forum.openwrt.org/viewtopic.php?pid=294828#p294828

So, is there possibly a patch for the driver necassary? :-/

I am totally new to Freifunk, OpenWrt and compiling Firmwares, but offer my best tries to help and test new versions on one of my devices.

Marcel

Procinger, 2015-10-20:
I think, we have to wait for an openwrt patch

https://dev.openwrt.org/ticket/20757

JamesT42, 2015-10-21:
I created that ticket and there has been no reaction to it yet. I think it would be OK to find a patch here instead of waiting for something to happen in an unknown amount of time in the future ;)

NeoRaider, 2015-10-21:
I've pushed a branch called 841v10 containing a (completely untested) patch that might fix the TX power issue. Don't forget to `make update` and `make clean` after switching to the branch if you want to test it.

Also, please send a dmesg of a boot with the patch applied.

NeoRaider, 2015-10-22:
The patch has been tested and is now included in the Gluon master.

closed 2015-10-22 by NeoRaider

marvin, 2015-11-01:
when i use the tag v2015.1.2 how can i use this patch in this tag then? thx for the help

NeoRaider, 2015-11-01:
@marvin, v2015.1.2 is based on Barrier Breaker, which has a too old version of ath9k to support the v10. You'll have to build the master, which will be released as Gluon v2015.2 soon (our current plan is to make the release by the end of November).

marvin, 2015-12-07:
any news on the release for v2015.2?