freifunk-gluon/gluon #520: WAN default routes leak into default routing table

NeoRaider, 2015-10-11:
Both 2015.1.2 and master are affected.

NeoRaider, 2015-10-12:
After further investigation I've found out that the route does not leak into the default routing table, but the ip rules cause table 1 to be inspected after the default table, even without fwmark 1.

NeoRaider, 2015-10-12:
I think we can't completely prevent the usage of table 1 as long as we use netifd for the address configuration (as the ip6table option moving the routes to table 1 also cases matching rules to be added).

I guess this bug is not really severe as the mesh default route should always be preferred. It will be fixed when we move to network namespaces.

jplitza, 2015-10-12:
Couldn't this be circumvented by stopping processing of the rules for packets without fwmark 1 after the rule that says "look in the main table"? Inserting the following in `/etc/config/network` after the existing `config rule6 'wan6_lookup'` should do:
```
config rule6 'main_lookup'
    option 'mark' '0x0/0x1'
    option 'lookup' 'main'

config rule6 'stop_fwmark0'
    option 'mark' '0x0/0x1'
    option 'goto' '4200000001'
```
(this actually doesn't stop processing but skips all the `lookup table 1` rules)

mweinelt, 2020-03-03:
I think this can be safely closed. I haven't seen any WAN route leaks on recent Gluon versions and table 1 is empty here.

closed 2020-03-03 by mweinelt