freifunk-gluon/gluon #524: autoupdater does not use the local dns server

mmalte, 2015-10-14:
If the fastd connection is broken and there is no local mesh the autoupdater fails to resolve any updateserver.

It would be great if the autoupdater would use the WAN DNS if the usage of the mesh DNS failed.

This would help to recover from firmware releases containing issues with fastd.

jplitza, 2015-10-14:
If the fastd connection is broken and there is no local (meaning WiFi?) mesh, how would the node reach the upgrade server? The connection is only established via mesh, not via WAN interface.

NeoRaider, 2015-10-14:
Gluon is designed to separate WAN and mesh strictly. The WAN interface is never used for anything but the fastd connection.

All normal services on the node, including the autoupdater, will only use the mesh for DNS and downloading the images. (At least it is supposed to work like that, there are some limitations to the separation that will be resolved in a future version of Gluon)

closed 2015-10-14 by NeoRaider

mmalte, 2015-10-14:
As long as I put an IP address into the site.conf the autoupdater is able to use the native wan connection. 

If I understood you right in your opinion it's the bug that the autoupdater uses the wan connection in this case. 

In my opinion the autoupdater should be able to use any possible path, for example to integrate a node which was turned off over a period of time in which the mesh protocol has changed.