freifunk-gluon/gluon #526: Support for TP-Link CPE210 v1.1

Sunz3r, 2015-10-15:
Hi!

I got this device yesterday. It seems to be identical to v1. So i have to modify the Hardware-ID in the Firmware-Header.
Im new in building luon firmware therefore i dont know how to add a patch proper.

In gluon/openwrt/tools/firmware-utils/src/tplink-safeloader.c
static const unsigned char cpe510_vendor[] = "\x00\x00\x00\x1f""CPE510(TP-LINK|UN|N300-5):1.1\r\n";
(...)
static const unsigned char cpe510_support_list[] =
        "\x00\x00\x00\xc8\x00\x00\x00\x00"
        "SupportList:\r\n"
        "CPE510(TP-LINK|UN|N300-5):1.0\r\n"
        "CPE520(TP-LINK|UN|N300-5):1.0\r\n"
        "CPE210(TP-LINK|UN|N300-2):1.0\r\n"
        "CPE210(TP-LINK|UN|N300-2):1.1\r\n"
        "CPE220(TP-LINK|UN|N300-2):1.0\r\n"
        "\r\n\xff";

NeoRaider, 2015-10-15:
Adding OpenWrt patches is simple: the `gluon/openwrt` directory is a git repository. You just commit your changes to this repository and then run `make update-patches` in the Gluon toplevel directory to create a patch file from it.

There are other places that have to be adjusted besides the image generator, I'll prepare a proper patch.

NeoRaider, 2015-10-15:
I've added a patch to support the new versions (the new image names need BROKEN=1 to be generated). Please check if the new devices are working correctly so we can remove the BROKEN check.

Sunz3r, 2015-10-17:
Thank you very much!
I test it with gluon-v2015.1.2 and it works perfectly. :)

NeoRaider, 2015-10-17:
@Sunz3r, so you ported the three firmware-utils patches to v2015.1.2?

Sunz3r, 2015-10-17:
Yes, i needed the last stable release. Could that be a problem?

NeoRaider, 2015-10-17:
No, I just wanted to make sure you tested the patches before I send them upstream.

closed 2015-10-17 by NeoRaider

drakessenyonga, 2016-05-25:
Hello all, Could you please firmware update file you used on cpe 210 v1.1. Will be grateful.
Thanks

FreifunkUFO, 2017-04-06:
so  @NeoRaider did you had time for a test? upstreaming would be nice.

NeoRaider, 2017-04-06:
@FreifunkUFO: CPE210/510 v1.1 support has been upstream since https://git.lede-project.org/?p=source.git;a=commit;h=21e1b4163f48a76dd34f35ca6e158f2b4e1aa03e ; since https://git.lede-project.org/?p=source.git;a=commit;h=d1a2c3f9b1f79baca2d16b8bf8bd6c409ee56e2e, also EU/US-specific variants are supported.