freifunk-gluon/gluon #529: luci webinterface on local port

A-Kasper, 2015-10-18:
I would like luci to always (or startable via commandline) run on a local port like 127.0.0.1:81
this would be nice to access the config mode via ssh tunnel. 
I didn't manage to start uhttpd with luci working. I always got: 

"No root node was registered, this usually happens if no module was installed.
Install luci-mod-admin-full and retry. If the module is already installed, try removing the /tmp/luci-indexcache file."

How is the config interface enabled? Any Ideas how to manage this?

NeoRaider, 2015-10-18:
The check `uci_state:get_first("gluon-setup-mode", "setup_mode", "running", "0") == "1"` in `/usr/lib/lua/luci/controller/gluon-config-mode/index.lua` in gluon-config-mode-core enabled the config mode LuCI module only when the node is booted into setup mode.

A-Kasper, 2015-10-18:
Am 18.10.2015 um 03:36 schrieb Matthias Schiffer:
> The check |uci_state:get_first("gluon-setup-mode", "setup_mode",
> "running", "0") == "1"| in
> |/usr/lib/lua/luci/controller/gluon-config-mode/index.lua| in
> gluon-config-mode-core enabled the config mode LuCI module only when the
> node is booted into setup mode.

I just want to use the luci webinterface to set the uci parameters. What
else does the config-mode do?
Should it work if I skip the test in config-mode/index.lua?

closed 2016-08-22 by rotanid