freifunk-gluon/gluon #532: Error while compile

freifunk-bamberg, 2015-10-27:
Traceback (most recent call last):
  File "../../tools/js2c.py", line 37, in <module>
    import bz2
ImportError: No module named bz2
make[6]: *** [/home/rene/statuspage/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/node-v0.12.7/out/Release/obj/gen/libraries.cc] Error 1
make[6]: Leaving directory `/home/rene/statuspage/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/node-v0.12.7/out'
make[5]: *** [node] Error 2
make[5]: Leaving directory `/home/rene/statuspage/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/node-v0.12.7'
make[4]: *** [/home/rene/statuspage/build/ar71xx-generic/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2_gluon-ar71xx-generic/node-v0.12.7/.built] Error 2
make[4]: Leaving directory `/home/rene/statuspage/packages/gluon/lang/node'
make[3]: *** [package/feeds/module_gluon/lang/node/host/compile] Error 2
make[3]: *** Waiting for unfinished jobs....
make[4]: Entering directory `/home/rene/statuspage/openwrt/package/kernel/linux'

NeoRaider, 2015-10-27:
This should not happen, OpenWrt brings its own python build which should contain the bz2 module; it might be a dependency issue. Could you please provide a full build log? You can use `make ... |& tee build.log` to write both stdout and stderr of the make command to build.log.

Subo1978, 2015-10-27:
the new statuspage need python with bz2 . i had the same problems on Ubuntu. python under ubuntu didn't support bz2. i tried all hint's but couldnt solve it.
I moved to archlinux. there it work's fine

aeickho, 2015-10-27:
on debian  8.2 

build.log
https://gist.github.com/aeickho/45dd8dbed75c1648554d

Delta1977, 2015-10-27:
for ubuntu and Debian i found no solution fur zhe bz2 problem  and moved to Archlinux.

freifunk-bamberg, 2015-10-27:
I will Do it  later in the evening

Rene Rössler
Administrator 
Radio Northsea International

<div>-------- Ursprüngliche Nachricht --------</div><div>Von: Matthias Schiffer <notifications@github.com> </div><div>Datum:27.10.2015  15:32  (GMT+01:00) </div><div>An: freifunk-gluon/gluon <gluon@noreply.github.com> </div><div>Cc: FF Weinstrasse <rene@radio-northsea.de> </div><div>Betreff: Re: [gluon] Error while compile (#532) </div><div>
</div>This should not happen, OpenWrt brings its own python build which should contain the bz2 module; it might be a dependency issue. Could you please provide a full build log? You can use make ... |& tee build.log to write both stdout and stderr of the make command to build.log.

NeoRaider, 2015-10-27:
Thanks for the information, I'll check in Debian myself and fix this.

NeoRaider, 2015-10-28:
This is fixed now.

After `make update`, it is probably necessary to run `make package/python/host/clean GLUON_TARGET=...` once to un-break the build, `make clean` is not enough.

closed 2015-10-28 by NeoRaider

freifunk-bamberg, 2015-10-28:
ok, thanks, i will test ist tomorow