freifunk-gluon/gluon #534: Kompiliert für TL-WR841Nv10, eine SSID namens OpenWrt.

Defab, 2015-10-28:
Moin,
Ich kann nicht ausschließen das ich irgendwo einen Fehler gemacht habe :)

commit war 3ddcf50b800cfdfca92400a5922ff1575ff3a8a9

git clone https://github.com/freifunk-gluon/gluon
make update
site.conf http://pastebin.com/YfxpKWKH
make -j5 GLUON_TARGET=ar71xx-generic

Jetzt habe ich 2 SSIDs, einmal "freifunk.hannover.net" zu der ich mich verbinden kann. Dann noch eine SSID "OpenWrt" zu der ich mich nicht verbinden kann.

cli show wireless
http://pastebin.com/GDD4T1d6

Ich habe die wireless config auf dem Router soweit angepasst das die SSID "OpenWrt" nicht mehr vorhanden ist.
Ich weiß jetzt nicht wo der Fehler liegt. Bei den v9ern die ich hier am laufen habe, habe ich keine SSID namens "OpenWrt".

Ich habe nochmal neu kompiliert mit
commit 172ad68134d91699144a6b762c2c8dbf2b5af518
allerdings ist der "Fehler" immernoch vorhanden.

A-Kasper, 2015-10-28:
Hast Du "Einstellungen behalten" ausgeschaltet beim Flashen? Evtl.
könntest Du noch Altlasten auf dem Gerät haben....

Defab, 2015-10-28:
Bei einen Router, den ich gestern schon geflashed habe, habe ich den Haken weggemacht. Die 2 Router, die ich Heute mit dem neuen Build geflashed habe, hatten noch die TP-Link stock FW drauf und die hatten eine solche Option nicht.

NeoRaider, 2015-10-28:
Also, ein paar Bemerkungen:

* Es kann definitiv nicht an der Hardware liegen, es muss etwas mit dem Flashen schieflaufen oder dem Image kaputt sein
* Erscheint eine Fehlermeldung, wenn du einmal per Hand `/lib/gluon/upgrade/200-wireless` ausführst?
* Tritt der Fehler noch auf, wenn du ein mit scp auf das Gerät kopiertes Image mit `sysupgrade -n /tmp/image.bin` flashst?

Könntest du ansonsten dein Image irgendwo hochladen? Dann schaue ich mal rein...

k3v1n-eu, 2015-10-28:
Hi,

wir haben in Euskirchen das selbe problem auf dem 841v10 mit der Master Version.
Unsere Beta images findest du hier: http://beta.ffeu.de/

Grüße
Kevin

NeoRaider, 2015-10-28:
@k3v1n-eu, seid ihr sicher, dass das Problem nur auf den 841v10 auftritt?

k3v1n-eu, 2015-10-28:
Bin leider noch nicht dazu gekommen, auf andere Router zu flashen. Werde es aber probieren und eine Rückmeldung geben.

NeoRaider, 2015-10-28:
@k3v1n-eu, eure site.conf ist noch gar nicht an den Gluon-Master angepasst, da kann die WLAN-Config so nicht funktionieren. Siehe http://gluon.readthedocs.org/en/latest/releases/v2015.2.html#site-changes

k3v1n-eu, 2015-10-28:
Damit hat sich das Problem dann erledigt. Ich denke das ist auch für @Defab die Lösung.
War vor dem Kompilieren zwar auf der Docu, aber war wohl Blind.

Danke für deine Zeit! :-)

NeoRaider, 2015-10-28:
Nein, bei @Defab sieht die Config eigentlich ok aus. In `wifi5` hat sich zwar ein Fehler eingeschlichen (`master` statt `ap`), aber das sollte keine Auswirkung haben.

k3v1n-eu, 2015-10-28:
Das ist mal schnell die geänderte von uns: https://gitlab.k3v1n.de/freifunkeu/site-ffeu/blob/v2015.2/site.conf

Delta1977, 2015-10-28:
@k3v1n-eu ist immer noch in der Sektion     fastd_mesh_vpn = {
        bandwidth_limit = {
        enabled = false,
        ingress = 3000,
        egress = 200,
        },
    },
falsch. hat sich auch geändert. schau dir mal im master unter doc's das an

NeoRaider, 2015-10-28:
Okay, I can reproduce this issue...

NeoRaider, 2015-10-28:
This broke in 22130e84e88618b8cfb93fe8fbff859c3e0115fa (two days ago), if you got a broken build, you need to flash a fixed version without keeping the configuration.

closed 2015-10-28 by NeoRaider

Defab, 2015-10-28:
Hab "master" in "ap" geändert und bin gerade am backen. danke fürs fixen.

commit 412e1ef1 (2015-10-28): gluon-core: fix removal of "OpenWrt" default WLAN interface

Fixes #534