freifunk-gluon/gluon #535: 841 v9 opkg install faild -> rebbot

aeickho, 2015-10-28:
with 172ad68134d91699144a6b762c2c8dbf2b5af518

i can't install openwrt packets ...

root@ffgoe30b5c222a464:~# opkg update
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/base/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_base.
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/base/Packages.sig.
Signature check passed.
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/packages/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_packages.
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/packages/Packages.sig.
Signature check passed.
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/luci/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_luci.
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/luci/Packages.sig.
Signature check passed.
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/routing/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_routing.
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/routing/Packages.sig.
Signature check passed.
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/telephony/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_telephony.
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/telephony/Packages.sig.
Signature check passed.
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/management/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_management.
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/management/Packages.sig.
Signature check passed.
root@ffgoe30b5c222a464:~# free
             total         used         free       shared      buffers
Mem:         28540        26592         1948          644         2120
-/+ buffers:              24472         4068
Swap:            0            0            0
root@ffgoe30b5c222a464:~# opkg install diffutils
Installing diffutils (3.3-1) to root...
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/packages/diffutils_3.3-1_ar71xx.ipk.

and ... 841 make an reboot

.....  openwrt.ffgoe is an reverse proxy for http://openwrt.draic.info/

aeickho, 2015-10-28:
no problem with TP-Link TL-WDR3600 v1

aeickho, 2015-10-28:
log on 841v9  with serial console

Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/management/Packages.sig.                     
Signature check passed.                                                                                                  
root@ffgoe30b5c2229f44:/# opkg install diffutils                                                                         
Installing diffutils (3.3-1) to root...                                                                                  
Downloading http://openwrt.ffgoe/chaos_calmer/15.05/ar71xx/generic/packages/packages/diffutils_3.3-1_ar71xx.ipk.         
[   67.900000] usign: page allocation failure: order:0, mode:0x20                                                        
[   67.900000] CPU: 0 PID: 2008 Comm: usign Not tainted 3.18.21 #2                                                       
[   67.900000] Stack : 00000000 00000000 00000000 00000000 803d4292 00000033 00000000 00000000                           
          00000000 00000000 80326e78 80380ba3 000007d8 803d34d0 807f6908 00000000                                        
          00000000 00000000 803807fc 800a388c 00000006 8008049c 00000000 00000000                                        
          8032a470 8049bc54 00000000 00000000 00000000 00000000 00000000 00000000                                        
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000                                        
          ...                                                                                                            
[   67.900000] Call Trace:                                                                                               
[   67.900000] [<800710f8>] show_stack+0x50/0x84                                                                         
[   67.900000] [<800cd110>] warn_alloc_failed+0xe8/0x10c                                                                 
[   67.900000] [<800cf000>] __alloc_pages_nodemask+0x51c/0x68c                                                           
[   67.900000] [<802319a4>] __netdev_alloc_frag+0x78/0x1c8                                                               
[   67.900000] [<80236e88>] __netdev_alloc_skb+0x5c/0x118                                                                
[   67.900000] [<80df8090>] ath_rxbuf_alloc+0x30/0x98 [ath]                                                              
[   67.900000] [<80ee69a0>] ath_rx_tasklet+0x588/0xafc [ath9k]                                                           
[   67.900000] [<80ee453c>] ath9k_tasklet+0x23c/0x2c8 [ath9k]                                                            
[   67.900000] [<80083194>] tasklet_action+0x84/0xcc                                                                     
[   67.900000] [<80082aa4>] __do_softirq+0xf8/0x230                                                                      
[   67.900000] [<80082e10>] irq_exit+0x54/0x70                                                                           
[   67.900000] [<80060830>] ret_from_irq+0x0/0x4

NeoRaider, 2015-10-28:
This is an out-of-memory issue. Sorry, this can't be solved... hardware with only 32MB RAM is already at its limit during normal operation.

closed 2015-10-28 by NeoRaider