freifunk-gluon/gluon #541: "Foreign" ARP table entries

Tobi042, 2015-11-01:
We observe, that there is an extreme number of ARP requests flooding our network from many sources. These are either by foreign networks or for foreign networks and by "foreign" I mean non-Freifunk networks, mostly private IP ranges such as the ones found in many home networks. 

I suggest creating an ebtables filter package that prevents such ARP packets (src IP does not match prefix4 from site.conf OR dst IP does not match site.conf) from being forwarded via batman. It might be useful to have them forwarded to other local devices, so the rule should probably not be applied at br-client level, but rather really at bat-device level.

tcatm, 2015-11-01:
It's valid usecase to use other IP Ranges (statically configured) within a layer 2 mesh network. I guess if such a feature is really desired the GCP might be a good place to implement it.

NeoRaider, 2015-11-01:
I think such a package would be pretty trivial, I don't really see an issue with make it an official package.

I think we can in fact block ARP/ND for foreign IP space without breaking anything useful. When other ranges are routed over the mesh network, the local nexthops should still always have "correct" addresses.

Tobi042, 2015-11-01:
I concede that it might be a valid use case, but I believe that most communities would profit from the reduced "management" traffic that such a package would ensue. As mentioned in my email on the mailing list, at times, Batman's distributed ARP table in our community is filled with 70% foreign IP addresses, with a non-negligible number of unresolvable ARP packets flooding every client regularly.
I could imagine an additional site.conf section that would define exceptions for such a rule. Also, normal clients would never see such out-of-range static IP addresses, because they would be L3-routed as the normal network subnet masks distributed by DHCP server do not allow for such traffic to be L2-delivered.

mweinelt, 2015-11-08:
Das scheint dieses Paket von Paderborn zu erledigen:
https://git.c3pb.de/freifunk-pb/ffho-packages/tree/master/ffho/ffho-ebtables-net-rules

PR #1015 (2017-04-09): [RFC] add package gluon-ebtables-source-filter
First draft to port ffho-ebtables-net-rules to Gluon

depends on #1016
fixes #541

decision needed:

- [x] the packagename is verry generic, so it should be changed to something that fits better, e.g. gluon-ebtables-strict-filter/-strict-net/-strict-source/-source-guard/-source-filter/...
  **=> gluon-ebtables-source-filter it is**
- [x] if no prefix4 is given within `site.conf` this package blocks all IPv4 traffic. Is that a reasonable default? **=> yes**
- [x] should we allow link-local-IPv4/APIPA/`169.254.0.0/16`? **=> no**
- [x] for documentat [... cut]
closed 2017-04-09 by NeoRaider