freifunk-gluon/gluon #543: Error while compile

freifunk-bamberg, 2015-11-02:
I become an aerror while i compile the master branch

here the build-log and site.conf:
https://github.com/freifunk-bamberg/bildlog

thanks

NeoRaider, 2015-11-02:
See https://github.com/freifunk-bamberg/bildlog/blob/master/build.log#L5626 of your build log: you don't have an autoupdater section in your site.conf, but some package in site.mk pulls in the autoupdater.

The site.conf error messages are usually quite readable, please try to make sense of them yourselves before opening a ticket...

closed 2015-11-02 by NeoRaider

freifunk-bamberg, 2015-11-02:
i have an section autoupdater in the site.conf

  autoupdater = {
    branch = 'stable',
    branches = {
      stable = {
        name = 'stable',
        mirrors = {'http://svc.services.ffws/fw/ffws/stable/sysupgrade/'},
        good_signatures = 1,
        pubkeys = {
                'db6d9cf00c46044dec849cb965e4ac8260a4f33b0f71b07ebff429a1a39ec195', -- Rene
        },
      },
      experimental = {
        name = 'experimental',
        mirrors = {'http://svc.services.ffws/fw/ffws/experimental/sysupgrade/'},
        good_signatures = 1,
        pubkeys = {
                'db6d9cf00c46044dec849cb965e4ac8260a4f33b0f71b07ebff429a1a39ec195', -- Rene
        },
      },
    },
  },

i see no error

NeoRaider, 2015-11-02:
Ah, the mistake is the lonely `next_node = {` in the middle of the file which completely messes up the structure. That mistake was in our documentation a while ago, so you probably copied it from there...

You should probably also improve your site.conf's indentation to make it more readable...

freifunk-bamberg, 2015-11-02:
yes, i found the error, now its compiling, thanks