freifunk-gluon/gluon #553: please review timing of MLDv2 for IPv6 / RFC3810

Joachim-Wolff, 2015-11-10:
Broadcast traffic for IPv6 Multicast Listener Discovery Version 2 floods the howl mesh network every Query Interval of currently 60 seconds (icmpv6.type == 143) . Every node has to answer to the request packet (icmpv6.type == 130) during the Query Response Interval.

The recommended timing from RFC3810 is:
Query Interval, default value: 125 seconds, gluon use 60 seconds
Query Response Interval, default value: 10000 (10 seconds) seams to be below 4 seconds in gluon, based on captured traffic on a node somewhere in the mesh. Sorry, haven't found the values in the source, but also don't know exactly where I have to search for...

[image]

36% of the packets arrive in the 1st second, 35% in the 2nd, 23% in the 3rd and the remaining 6% within the 4rd second. Makes a peek of 350 pkt/s with ~970 nodes in the mesh, expecting each one has to be handled by the IPv6 stack... Maybe this explain the high sirq values or the load issue #446

Joachim-Wolff, 2015-11-11:
My nodes have answered all request faster than 30ms, only one exception with 65ms, so without delay...

Expecting every node do so and there is no large delay in transmitting the request on uncongested lines, there must be a fine broadcast peak. The visible time in the graph seams to be the transmitting time through the mesh, and every packet must be stored somewhere for this time, because it's broadcast and all packets are delivered successfully. Fortunately the packet size is only 158 byte :)

tcatm, 2015-11-11:
You may have a misconception about how MLDv2 works. It's up to the querier to decide on both intervals and (If I'm not mistaken here) any host on a layer2 mesh may become a querier (even a Client). Could you check which host has taken that role in your mesh and whether it is running Gluon at all?

Joachim-Wolff, 2015-11-11:
Sorry, have read only 30% of that RFC, and they always use the wording router...

But you're right, I've captured on 5 days and it always was a client (not the same, maybe smartphone) behind a Freifunk-Router that send the queries, with a Maximum Response Time of 10ms, so all works like designed... :(

So maybe change the title to "take control over MLDv2" ??
How handle other communities this issue, and are they aware of it?

T-X, 2015-11-11:
Do you have a MAC address of the querier? I'd be interested in the first three bytes, the vendor part, to know what kind of device is sending these weird queries. Maximum Response Time of 10ms seems like a bug indeed.

Lübeck or Hamburg have queriers in their networks. Last time I checked they seemed to be some enterprise switches connected to the LAN ports of a gluon router. But they had more sane response intervals, so they weren't an issue.

A short-term option would be to place your own querier in the network with lower intervals and a lower mac address to win the election.

#402 includes filter rules to split the IGMP/MLD domain on a per node basis. They could be applied without the other patches in that pull request.

Joachim-Wolff, 2015-11-11:
Winstars 80:3f:5d:xx:xx:xx  9.nov
Tecnomen 00:e0:20:xx:xx:xx 10.-15.oct.
but I only captured sometimes.

I'm located in Hamburg, but currently it seam to be calm...

ohrensessel, 2015-11-18:
My experience from Hamburg is the following: during some times we had an LG device in the mesh (a TV?) sending out multiple MLDv2 packets per second.

closed 2016-03-08 by NeoRaider