freifunk-gluon/gluon #555: TP-Link 841n V10 + 940nV3 + 941ndV6 have less transmitting power (dbm) than specification?

KK4, 2015-11-11:
Ih tested three TP-Link 841n V10 + 940nV3 + 941ndV6, and all have 18 dbm only as max
also,
In LuCI the v9 has range "standard" , 19 dbm = 79 mW, 18 ... and so on 
the V10 has "standard", 18 dbm = 63 mW , 17 ... and so on  (19 dbm is missing)
the 940/941 have "standard", 20 dbm = 100 mw, 19 dbm ... and so on 
I think, "standard" should be the next step above always.

However, all these give effective 18 dbm = 63 mW only, even if I set "standard" or the next max

I tested this with bit, the orginal firmware as well as with the last  FF firmware, it is same. (teh original firmware has three settings only, "high", "middle", "low".

I asked TP-Link technical support
Matthias Korn
Technical Support Engineer 
and got (for the 841n V10 only, for 940/041 they could nit provide the specifications ) the answer below:

"hier ist kein Fehler vorhanden sondern alles völlig normal"

"Laut den Spezifikationen des TL-WR841N(UN) Version 9.0 betragen die maximal gemessenen dbm Werte 20dbm für 11b, 21dbm für 11g und 20dbm für 11n.

Beim TL-WR841N(UN) Version 10.0 betragen diese Werte 19dbm für 11b, 21dbm für 11g und 21dbm für 11n. "

die Spezifikationen für ihre Hardware Versionen von TL-WR940N und TL-WR941ND habe ich nicht vorliegen.

NeoRaider, 2015-11-12:
This is correct. AFAIR, the limits are obtained by the WLAN driver from the ART partition (radio calibration data) from the flash.

The value "default" simply means that we don't set the limit at all, but use the default setting from the WLAN driver. This is usually the same as the maximum value displayed in the list, but we don't know for sure in the config mode, as the WLAN is not active then. I guess we'll look at the code to find out how the default value is determined eventually, so the "default" entry can be replaced by the correct value in the future, but it doesn't have very high priority as it is only a cosmetic issue.

Furthermore, it is unclear what these values mean at all. At which point in the transmission chain are these values measured? Do they contain the antenna gain? (Probably not) Wire loss? External amplifiers? (Probably don't exist on typical TP-LINK hardware).

It is probable that the limits are set in a way that ensures the regulations are observed with the default antennas. I think the default antennas have 3dBi, so a limit of 18dBm combined with some loss would yield a total of about 20dBm. In my opinion, this isn't really anything to care about. A difference of 2-3dB usually doesn't have a noticable effect in practice.

A better framework which could allow the user to set the antenna gain themselves if they have replaced the antennas, but still have meaningful defaults,  would be nice. At the moment, the antenna gain is not used at all to determine the default TX power, leading to much too high TX power on some devices (see #94).

By the way, such questions are better fit for the Gluon mailing list (or some other related MLs like wlanware and OpenWrt).

oscarminus, 2015-11-30:
It seems that this is not only a problem of the models mentioned above. I tested the firmware based on gluon master with a 841 v8 and v9. Both models have the same issue: Running Gluon 2015.1.2, the transmit power is set to 18dBm on the v8 model and 19 dBm on the v9 model. After upgrading to gluon master, iwinfo reports 16 dBm as transmit power on the v8 and the v9 model. 

```
root@grover:~# iwinfo
client0   ESSID: "paderborn.freifunk.net"
          Access Point: C2:4C:01:A0:57:52
          Mode: Master  Channel: 1 (2.412 GHz)
          Tx-Power: 16 dBm  Link Quality: unknown/70
```
iwinfo says that power could be set to 18dBm on this model but even setting this by hand does not work.
```
root@grover:~# iwinfo client0 txpowerlist
[...]
  14 dBm (  25 mW)
  15 dBm (  31 mW)
* 16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
```

Actually the lost 3dBm cause that one of the worse mesh connections fails. Before it was not the best, but it was usable. So 3 dBm does matter.

Delta1977, 2015-11-30:
The same here on an TP-Link 841 v9 :

root@nodename:~# iwinfo mesh0 txpoewerlist
……..
  15 dBm (  31 mW)
* 16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)
root@nodename:~# iwinfo client0 txpoewerlist
………
  14 dBm (  25 mW)
  15 dBm (  31 mW)
* 16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
  20 dBm ( 100 mW)

greets

Amtsgericht Köln HRA 17574

Pers. Haft. Gesellschafterin:

MAXIM Markenprodukte Verwaltungs GmbH

Sitz: Pulheim

Amtsgericht Köln HRB 40643

Geschäftsführer: Dr. Rolf Giesen

SenorKaffee, 2015-12-01:
Checked with 841v8 (gluon-v2015.1-287-g6110d3d) and noticed the same thing, it's no longer running on max by default.

root@ff-re-Buddestr-Kuechenradio:~#  iwinfo client0 txpowerlist
[...]
  14 dBm (  25 mW)
  15 dBm (  31 mW)
* 16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)

39mW and 63mW seems to be a pretty large difference.

A-Kasper, 2015-12-01:
Am 01.12.2015 um 08:08 schrieb Michael Herzog:
> Checked with 841v8 (gluon-v2015.1-287-g6110d3d) and noticed the same
> thing, it's no longer running on max by default.
>
> root@ff-re-Buddestr-Kuechenradio:~# iwinfo client0 txpowerlist
> [...]
> 14 dBm ( [...]

in addition:

I've got two 1043 v2 with different max txpower values.
I didn't handle to match them.

A-Kasper, 2015-12-07:
https://dev.openwrt.org/browser/trunk/package/kernel/mac80211/patches
I think the new patches should solve this. I'm not sure if gluon uses or ignores these patches. 
The most interesting ones are 402 and 404
402 forces the driver to accept the user settings instead of the ones in eeprom
404 disables analysing the neighbourhood for the correct regdomain
I found these patches while searching for reasons, why the txpower under CC is restricted. The first attemp to fix it I found was: https://patchwork.ozlabs.org/patch/444666/ 
maybe this helps to fix the issues

NeoRaider, 2015-12-07:
@A-Kasper, these patches are really old (I can't remember an OpenWrt version without them), and of course, Gluon contains them as well.

ATH_USER_REGD is already enabled by Gluon, so https://patchwork.ozlabs.org/patch/444666/ doesn't help either (without ATH_USER_REGD, we wouldn't be able to change the regdom from US to DE).

Looking at the OpenWrt code doesn't necessarily help much, the change might also be in the kernel. I'll try to bisect this, anything else is just a waste of time.

discsimo, 2015-12-12:
You can solve this problem manually with the following steps (tried it on my TP-Link TL-WR841N v10, gluon-v2015.1-262-g77829cd):

•	uci set wireless.radio0.country=00
•	uci set wireless.radio0.txpower=21
•	uci commit
•	reboot
Afterwards  iwinfo client0 txpowerlist looks like this:
 [...]
16 dBm ( 39 mW)
17 dBm ( 50 mW)
*18 dBm ( 63 mW)

/etc/config/wireless contains also the correct settings.
Seems not to be a generell driver problem, only with country code "DE" maximum txpower is set to 16dbm. 
Maybe this helps to fix the issue.

rotanid, 2015-12-12:
@discsimo and you are sure you don't violate german law with that increased output power?

KK4, 2015-12-12:
I think it is according German law, what limits to 20dbm.
There is no need to limit the txpower to 16 dbm

I compared the effective power output with original antennas for 841n v10, 1043nd v2 (current stabile firmware), 940nd v6 and then they all have same = 20dbm (with original tp-link firrmware it powers the same, exact 20dbm)

This is independent if the table goes until 18 only, or to 20, and what the asterix says or if there is no asterix.

Also, after that modification I checked country settings and as active with German figures and 20dbm.
Looks for me like a bug causing confusion.

A-Kasper, 2015-12-12:
you need to add 5db antenna gain. so the limit with standard antennas is
correct. but we use some setups with long cable so the poer should be
configurable as needed

KK4, 2015-12-12:
don't think this is right so, because
1) they have different antenna gains, but restsriction is same for all
2) they don't have 5dbm antenna gain with standard antennas, from where yoy have these figures?
3) *If* you are right, this would mean, that tp-links deliveres all routerss with wrong settings for DE, as the effective dbm is 20 with its antennas.
4) *If" you are right, then all (also all installed 30.000 routers) are not according German law and need to get modifies, what means all will have about 50% less distance for meshes, 
5) *If* you are right, same is for Fritzbox and other routers. I simply cannot (will not) believe, that all these sold and installed routers are not correct working an until now no one complained that and forces a country wide update for all. This is not logic and against all life experience.

Marssl78, 2015-12-13:
I don't know if this has anything to mean in here, but I have an 841v10 running openwrt. I yoused a version of openwrt, somebody has built and put in the forums, which contains the tx-power-fix. In the Luci tx-power menu, the pulldown list only offers 18db when i select channel 1 as wlan channel. when i switch to channel 7, the pulldown list offers even 20db. Does this give a hint to you or is it even more confusing? :-/

KK4, 2015-12-13:
I tested the experimental from PetaByteBoy made on this base, now as sysupgrade for the 841v9 to see wht it will do with the old versions.

Result is terrible: 16dbm is hardcored, even with the workaround above it always is restrictedd to 16dbm = 39mW

root@841n_v9_a:~# iwinfo wlan0 txpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
   2 dBm (   1 mW)
   3 dBm (   1 mW)
   4 dBm (   2 mW)
   5 dBm (   3 mW)
   6 dBm (   3 mW)
   7 dBm (   5 mW)
   8 dBm (   6 mW)
   9 dBm (   7 mW)
  10 dBm (  10 mW)
  11 dBm (  12 mW)
  12 dBm (  15 mW)
  13 dBm (  19 mW)
  14 dBm (  25 mW)
  15 dBm (  31 mW)
* 16 dBm (  39 mW)
  17 dBm (  50 mW)
  18 dBm (  63 mW)
  19 dBm (  79 mW)
root@841n_v9_a:~# iw reg get
country DE: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS
        (5470 - 5725 @ 160), (N/A, 27), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
[image]

KK4, 2015-12-15:
I come back to  A-Kasper "you need to add 5db antenna gain."

I tolked with tp-link, and they confirmed, that the EIRP CE: <20dBm(2.4GHz) *inclusive standard antennas.
Means, you have to add nothing. It is a common misundertanding, that the specification are w/o antenna if the antenna gain is told in additon.

I also refer also to http://www.com-magazin.de/tipps-tricks/wlan/wlan-strahlung-zulaessige-grenzwerte-920715.html (in German) "...Diese Angaben beziehen sich auf die *Strahlungsleistung des gesamten Geräts inklusive Antennen*. Die Strahlungsleistung als EIRP (Equivalent Isotropically Radiated Power) berechnet sich wie folgt: elektrische Sendeleistung des Access-Points (in dBm) plus Antennengewinn (in dBi) minus Dämpfung durch Antennenkabel, Stecker und so weiter (in dB)..."

Means, as all the tp-link models delivered in germany have CE: <20dBm(2.4GHz they are specified as 20dbm maximal incl. antenas as delivered.

Furthermore, I made a comparison with Fritzbox 7360, and they have exact same total power like tp-link 841n v10, 950n v3 in original, and same if  take the current experimental firmware and lift the txpower=20

So I believe, those who are responsible for this restriction to 16dbm simply made a fallacy, and never tested and carry out measurements with the hardwarte in case and compared it.

If someone does use other antennas, it is his responsibility to lower the power, that is not the programmers responsibility..

A-Kasper, 2015-12-16:
Hoe exactly, does the chip know which Antennas are connected? It's a common
SoC which is used in a lot of devices. The driver can't detect the antenna
gain. That's the problem. OpenWRT wants to stay legal, so I think they set
the maxvalue for the controller txpower to 16 dB.

The relevant EIRP Power is:
Controller Outputpower + Antenna Gain - Cable-Loss

In my mind we can just set the Controller Outputpower by Software. If you
set it to 16 dB and add the 5 dB a-gain you'll get the legal 100mW.

If the value you set in the driver the antennagain have to be stored in the
controller flash. That's possible, but I don't think so. I guess, that the
txpower in the old firmware was to high and ignored the Antennagain.

KK4, 2015-12-16:
this is the fallacy, exactly.
> does the chip know which Antennas are connected? It's a common SoC which is used in a lot of devices. The driver can't detect the antenna gain
Ther chip does not need to know it. The point is the software setting by country (and its limittions) This the chip reads. Ans as the specification for the routers say EIRP <20dbm with original antennas (wich are 2 x 5dbm for 841 and 3x3 dbm for 950) it sets the ampere (x volt) by this to have just that limit gives by the country definitions. 
In original the hardware gives with setting DE (by software), before antennas a physically output of 9dbm, plus antennas = 19 (for the 841 with 2x5dbm), resp. 11-12 for the 940 with 3x3 dbm plus antennas = 20 .
As you say
>The relevant EIRP Power is: Controller Outputpower + Antenna Gain - Cable-Loss
yes, this is *inclusive* antenna gain ! (for the original <20 dbm)

The versions up to v9 all have it correctly, and ther effective output is same as the output of the original routers..

If you don't believe it carry out measurements with the hardware. Routers with the new experimental firmwares have ~60% of power only compared with original factory hardware.
Means, you run into a fallacy

NeoRaider, 2015-12-16:
I think this discussion is not really getting forward, and there are some misunderstandings.

@KK4, you are right, the transmission power *relevant for the regulations* does include the antenna gain. I don't think there were any misunderstandings here. The basic question was if the TX power *displayed by OpenWrt* does include the antenna gain, and we're still not sure if it does. These are the possible explanations:

1. If does -> The displayed value will only be correct with the stock antennas, as the router can't know that different antennas are used
2. It doesn't -> The 20dBm that were shows until Barrier Breaker were much too high, and the 16dBm on Chaos Calmer still are (?).
(3. It contains a random offset?)

The only way to know what the values displayed on OpenWrt actually mean is to measure the actual output with professional equipment and compare it with the stock firmware.

As someone has mentioned in this thread, the 20dBm that Barrier Breaker is showing seem to be the same as the maximum TX power of the stock firmware, and the 16dBm on Chaos Calmer are really 4dB less (and it is not just a display issue). If these measurements are correct, this would mean that explanation 1 is correct.

KK4, 2015-12-17:
you can do a simple comparison: original stock tp-link with setting DE
gives same result as experimental with txpower=20 + country=00

[image]

[image]

CodeFetch, 2016-01-14:
@KK4 Did you really have to open up this discussion?

I did measurements to compare the 1043ND v2 and 940N v3 a while ago with US-regdom.
The 940N has a MUCH higher maximum EIRP than the 1043ND.
The reason for that might be that the TP9343 built into the 940N was actually developed for the chinese mainland.

> In original the hardware gives with setting DE (by software), before antennas a physically output of 9dbm, plus antennas = 19 (for the 841 with 2x5dbm), resp. 11-12 for the 940 with 3x3 dbm plus antennas = 20 .

That is wrong and please don't call antenna gain "dBm". It's "dBi".

> I tolked with tp-link, and they confirmed, that the EIRP CE: <20dBm(2.4GHz) *inclusive standard antennas.

In their definition of "EIRP"...

Actually I have never seen a N wifi router that observes restrictions of the Bundesnetzagentur and I never heard of the Bundesnetzagentur measuring EIRP of wifi devices.
Do I really have to go into detail?

I think the Gluon developers shouldn't need to worry about things that even the original distributors don't when it comes to regulations.

NeoRaider, 2016-01-14:
If the Gluon master provides lower TX power than older versions, and than the stock firmware, we should fix that. I don't really care too much if TP-Link's measurements are accurate.

There's a patch now which might affect the issue: http://git.openwrt.org/?p=openwrt.git;a=commit;h=3550e5b2ec6bc49c2b73b00989a5e79fa9860c6c I'll check if it fixes the issue.

NeoRaider, 2016-01-28:
I've bisected the change to http://git.openwrt.org/?p=openwrt.git;a=commit;h=37a90b43a1d8f9be05ca461e700a59c844715740 ; AFAICT, this is not a bug, but an intentional change.

I've also tried to measure the TX power on the reported devices. On the TL-WR841N v9 and v10, new TX power (which is reported as 16 dBm on OpenWrt) seems to be the same as used the stock firmware when set to German regdom; this leads me to the conclusion that the new value is correct, and the old one was too high.

For the TL-WR940N v3, the TX power seems to be a bit lower in OpenWrt than in the stock firmware, but only by about 2-3 dB. OpenWrt reports 14 dBm here.

That said, I have to admit that my test setup wasn't very professional, I just used my Android phone with the app Wifi Analyzer to measure the power. Occasional disturbances had a much higher impact than the differences I tried to measure (sometimes about 5 dB), so everything has to be taken with a grain of salt. I've tried my best to get reproducible results by measuring for 5-10 minutes and repeating the measurements a few times.

viisauksena, 2016-01-30:
also ein paar Tests aus Freiburg dazu : 
setting: 2 Router beide mit 2016.1 (master,latest fw und der neuen Statuspage und neighbour)

ssh login auf testrouter und kontrolle über statuspage 2. router! (2. Router 841N v9 und später mit gleichem Ergebnis noch ein 710Nv1)

Kontroll settings;  
**iw client0 set txpower fixed 100 ; sleep 1 ; iw client0 set txpower fixed 2000 ; sleep 1 ; iw client0 set txpower fixed 1000 ; sleep 1; iw client0 set txpower fi
xed 2000**
liefert schöne treppen und läst sich beliebig verfeinern in der Ausgabe der Statuspage auf den Kontrollroutern

Am Testrouter:
wechsel der regd : Anzeige bei (iw reg set DE) db -> 16 und bei (iw reg set 00) db -> 19
kontrolle: **iwinfo |grep Tx** 
Wirkung auf den 2. Router in der Db Anzeige keiner
Gleicher Test wie zuvor verlief gleichbleibend, also wechsel der txpower, liefert instant schöne treppchen.

Bedeutet für mich, ja in DE (regd) steht da jetzt 16db, das sind aber sonst auch immerschon 16 db gewesen, nur das die da die Antennengewinn mit reingerechnet haben.
Die 3 Db hätten schwach aber sichtbar sein müssen, empfang Router 1 -35 Db , Router 2 -50 Db

DISTRIB_REVISION='r47335'
DISTRIB_CODENAME='chaos_calmer'
DISTRIB_TARGET='ar71xx/generic'
DISTRIB_DESCRIPTION='OpenWrt Chaos Calmer 15.05'
TP 841 N v9 - normale Antennen am Testrouter.

hoffe das hilft mal zur klärung - mir hats gehoöfen, hab mir schon sorgen gemacht.

dominikhlbg, 2016-02-01:
@NeoRaider It's not correct. The problem already exists in your link. The lower tx power problem began in patch Bleeding Edge, r43032 at the same day of commit. http://git.openwrt.org/?p=openwrt.git;a=commit;h=0e82759adb84c5c5fc9c78243422446c550acd19

KK4, 2016-02-01:
ein anderer Vergleich wäre einfacher: das Orginalgerät ist sicherlich nicht höher, als die Betimmugnen erlauben.
Ich hab noch mal mit TP-Link Support telefoniert, und die sagen:
- >Spezfikation nennt Wert incl. Antennengewinn und "Standard" ist immer Maximalleistung, unabhängig welches Land immer die selbe, weil die Ländereinstellung nur die Frequenzwahl nicht aber die Leitung steuert (bei der Originalfirmware) . Ein Drosseln ist also unnötig, Hardware-Maximalleistung ist mit Antennengewinn sowieso niedriger.

Also einfach vergleichen: ein Originalgerät und ein 841v10 mit neuem gluon.
Wenn beide dieselbe Leistung haben, ist alles OK, ist das 841v10 mit neuem Guol schwächer, ist was unnötig gedrosselt.
Vergleichergebnis: mit neuem Guon ist das 841v10 mehr als 50% schwächer, (ein db = Verdoppelung der Leistung) und mit Änderung auf Land = 00 ist die Leistung identisch.

Das alte 841v9 bringt incl. Antennengewinn 18, das neue 19, erlaubt wären 20 = 100 mW

A-Kasper, 2016-02-01:
> (ein db = Verdoppelung der Leistung)

3dB  sind ungefähr eine Verdopellung der Leistung.
Ich sehe es aber auch so, dass die Bremse wieder raus muss. "Nur 3dB
Unterschied" sind eben 50% Leistung.

Adorfer, 2016-02-05:
Es hat für mich den Eindruck, dass die Ermittlung der zulässigen maximum-txpower als "kleinster gemeinsamer Nenner" gefahren wird. 
Wenn man erstmal die Regdomain, den Kanal **und** den HT-Mode richtig setzt, die Initialisierung abwartet und dann nochmal _iwinfo radio0 txpowerlist_ holt, dann gibt es sowohl für den 841v9 als auch für den 841v10 (zumindest auf den unteren Kanälen) wieder 20dB als zulässigen Einstellwert.

viisauksena, 2016-02-09:
@Adorfer ich dachte ht-mode support is dropped ( http://gluon.readthedocs.org/en/latest/releases/v2016.1.html ) Kanal und Regdomain über die site.conf sind nicht ausreichend, oder meinst du das man das trotzdem noch setzen soll?

Adorfer, 2016-02-09:
@viisauksena Ich verstehe den Zusammenhang nicht. 
Ausserdem denke ich, dass die Frage der korrekten TX-Powerlist nicht Thema dieses Issues ist. 
Hier geht es ja um die fehlende Leistungstufen "generell" und nicht nur das Problem vom Setup-Mode.

viisauksena, 2016-02-09:
@Adorfer .. ich beziehe mich auf deine Aussage "Wenn man erstmal die Regdomain, den Kanal und den HT-Mode richtig setzt, die Initialisierung abwartet"
was / wie meinst denn du das das jeweils umgesetzt wird am einzelnen router wenn nicht durch die FW und damit die site.conf

Adorfer, 2016-02-09:
@viisauksena 
Dann war meine Aussage falsch und ich bin hier im falschen Issue, da es hier offensichtich um Openwrt-Probleme geht und nicht um Gluon-Issues. 
Mein Anliegen ist daher (jetzt) in #641

SimJoSt, 2016-04-14:
After updating a TP-LINK WDR3600 I noticed, that it had reduced transmission power.
My CPE210 with the new status page showed it with -85dbm - -90dbm before the update. After the update it wasn't shown at all.
I cannot contribute anything to whether it was set on a too high value before or is now beneath the limit.

I will perform any requested tests, if possible.

RubenKelevra, 2016-08-12:
We just performed a throughput test with two freifunk-nodes, we used one TL-WDR4300 and one TL-WDR3600 and placed them in the following three setups:

- 1m in sight
- 3m with one wall in between
- 10m with three walls in between

We used two VeEX VePAL TX300s to transmit 50 mbit/s in one direction and 1 mbit/s in the other via one continuous stream. We used L2 ethernet packets with 1518 bytes length.

The results are the following for gluon 2014.4 (IBSS):

- 1m in sight: **< 0.00% packetloss**
- 3m with one wall in between: **< 0.05% packetloss**
- 10m with three walls in between: ** 3 to 6% packetloss**

The results for gluon with the latest stable (2016.1.x) with the following results (IBSS):

- 1m in sight: **< 0.00% packetloss**
- 3m with one wall in between: **< 0.00% packetloss**
- 10m with three walls in between: **~ 0.1% packetloss**

To push it to the limits, we increased the sending speed to 70 mbit/s instead of 50 mbit/s
- 10m with three walls in between: **< 1% packetloss**

----

Conclusion: 

We learned that the newer firmware performed much better, in our setup. We are open for suggestions for further tests if desired.

[image]

SimJoSt, 2016-08-15:
Nice benchmark, interesting results.
In order to adress the core of this issue, a maximum range test with both firmwares would be much appreciated. How far can the routers be apart until they don't link/mesh anymore, don't see each other anymore. My interest would be satisfied with results taken from the statuspage.
The setup doesn't matter to me as well. Outside or with a lot of walls, I just want to know if the new firmware stops linking earlier.

RubenKelevra, 2016-08-15:
@SimJoSt what do you think about an dBm-test on the antenna-connectors? That would be much more comparable and reproducible.

The overall range coverage is much reduced on our newer firmware, we have already the b-disable patch included for a year now: https://github.com/freifunk-gluon/gluon/pull/810

So our "old" firmware does not include it, but the new one does.

We don't think this change anything on this setup, since we stayed well above the basic_rates - in the extended rates - which are changed with the patch. 

So in the new firmware does not get a that huge coverage but the links are much more usable if connected and the overall mesh is improved, because the hotspots does not try to send with very low end transmit rates to get some meters more range.

Best regards

Ruben

SimJoSt, 2016-08-17:
Oh, I didn't know that. Thanks!
Of course a dBm test would be much more perfect.

RubenKelevra, 2016-08-17:
Actually a dBm test seems to be hard as well, cause we use TPC per packet now. But I think the beacons are still send with maximum txpower. So I think we can measure them with 1 Mbit on new/old firmware. 

The txpower is usually different on different datarates so with actual data it's hard to measure. I think we'll borrow some hardware if available and give it a try. :)

//cc @bitnukl

NeoRaider, 2016-09-08:
I have just verified that the maximum TX power of 16dBm seen on the TL-WR841 v10 (current LEDE, country DE) is the same power the stock firmware uses on level "high" when country is set to Germany.

Verified using devmem:
```
# devmem 0x181080e8
0x3F3F1818
```
The returned value is the same on the stock firmware and LEDE. I expect that tests on other devices will yield similar results.

Thus, I conclude that the new behaviour of OpenWrt/LEDE is correct and this is not a bug.

closed 2016-09-08 by NeoRaider

RubenKelevra, 2016-09-09:
@NeoRaider thanks for Pointing this out :)

jyotirajsharma, 2017-09-14:
I have TP-Link 841n (V10) router. To run linux command, do I need to install OpenWrt firmware again or I can run linux command directly connecting router via serial console??