freifunk-gluon/gluon #556: TL-WR941ND v6 continuously rebooting when meshing via 802.11s

kpanic23, 2015-11-13:
I have flashed a 941 v6 with gluon based on newest master.
When there's another router using 802.11s in reach, the router starts rebooting.
It seems to run quite solid when only meshing via IBSS. As soon as I update one of the other nodes in the vicinity to master, the reboots start to happen.
I have connected a serial console to the 941 and captured the console.
Maybe this might help someone in getting the 941 v6 running stable...
[minicom.cap.txt](https://github.com/freifunk-gluon/gluon/files/33724/minicom.cap.txt)

NeoRaider, 2015-11-13:
This is a out-of-memory issue. Most of our devices are just too weak to run IBSS and 11s at the same time, you can only use one or the other, but not both.

Does the issue also occur when you disable IBSS meshing? If it does, there might also be a memory leak in the 11s code.

This is definitely not a hardware-specific issue, the backtraces only show generic code.

NeoRaider, 2016-02-05:
Closing for lack of feedback.

closed 2016-02-05 by NeoRaider