freifunk-gluon/gluon #557: Fastd keeps old peers when updating to firmware with new site.com

mmalte, 2015-11-13:
Since we had two fastd connetions per node I added peergroups in the past to force the nodes to connect to gateways in different data centers.

Since we changed the peer limit to one and started to move the gateways around I like to remove the peergroups.

Sadly the old entries are not removed from the /etc/config/fastd during autoupdate.

jplitza, 2015-11-13:
Yes, this is intentional to allow custom fastd peers to be added.

You can, however, remove old peers by specifying a peer group with the
old name but without any peers in your site.conf. There will still be an
empty peer group lingering, but that shouldn't do any harm.

(For details, have a look at the code:
https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-mesh-vpn-fastd/files/lib/gluon/upgrade/400-mesh-vpn-fastd#L83)

mmalte, 2015-11-13:
I feared that this in intentional.

Is this a historic decision or is it still considered useful?

This feature creatures a lot of trouble when migrating nodes from one community to another and forces me to keep an old structure in my site.conf forever.

If there is anybody using this feature at all, I am absolutely sure that person is able to do manual upgrades or build custom firmware.

A-Kasper, 2015-11-13:
The remaining peer config is a problem. I've had various bridges to
Bochum (Hamburg, Mönchengladbach) because the users don't use the
"delete all config" button. There should be a "keep peers" line in
site.conf if someone really wants to keep the peers. maybe the update
script could delete old fastd config before flashing

tcatm, 2015-11-13:
As all desired features are already implemented I guess we should improve documentation a little so people can use them correctly.

A-Kasper, 2015-11-13:
Am 13.11.2015 um 16:12 schrieb Nils Schneider:
> As all desired features are already implemented I guess we should
> improve documentation a little so people can use them correctly.

It's not a documentation problem. It's impossible for me to enter all
possible other peernames/groups in my site.conf to force them to be
deleted. If someone from another domain flashes a bochum image i can't
do anything.

The problem comes with nodes from other domains. Community intern I'm
able to read and use it the correct way.

mmalte, 2015-11-13:
>It's not a documentation problem. 

I have to support that. 

I'm not able to remove old entries which are no longer needed. 

Keeping empty dummies is just a workaround.

NeoRaider, 2015-11-13:
When firmwares are upgraded to different sites, pretty much all config can break when it is not reset; adding a hack for one single setting does not help much. We aren't really happy with the state of the configuration management of Gluon either, but this must be fixed with a major overhaul (maybe completely dropping UCI, we're not sure yet...), not small fixes. Keeping dummy entries may be a workaround, but it works for now.

Using custom VPN peerings instead of connecting centralized servers should be encouraged. This is not really practical with batman-adv as it has too high overhead, but with layer 3 routing protocols it should be much more realistic.

Adorfer, 2015-11-22:
@mmalte How many different dummy-groups of other communities do you want to maintain? 
Perhaps these peergroups should be documentented in the freifunk.api, in order to keep track of all other possible peergroups (and to have those as empty in your own site.conf)

NeoRaider, 2015-11-22:
@Adorfer, as I wrote before, this is not going to help. There are other config files that will potentially break when unrelated communities' images are flashed without clearing the config (for reasons like incompatible package choices, or different Gluon versions). Just adding a workaround for the fastd config is making the situation worse, as the breakage becomes even harder to detect.

One possible solution would be the following: site.conf contains a list of "supported upgrade sites" from which an upgrade is possible cleanly. If the upgrade scripts detect that an unsupported upgrade has been attempted (from an unknown site config, or a downgrade), it will lock the device in a "reset mode" that will only allow resetting the configuration.

For devices with overlay, `firstboot` could also be called automatically in these cases, so the upgrade will just reset the config when keeping it is not supported, but on targets without overlay (i.e. x86), resetting the config is only possible by flashing a new image. But even on devices with overlay, having the "reset mode" might be preferable so the user knows what is going on.

A-Kasper, 2015-11-22:
maybe we should look for workarouds. some ideas:
- a site.conf peer naming convention could be defined and forced by
pre-checking it
- all old fastd peers could be deleted, if communitystring changes
- all old peers could be deleted during update process by default. a
uci/lua setting "keep peers" could change this behaviour
- Custom Peers in extra section. only custom peers survive updates

mmalte, 2015-11-22:
@Adorfer These are just or own deprecated peergroups from the time when we hat 2 tunnels per node.

viisauksena, 2016-01-22:
i encounter that the order of the dummy is pretty important, to have a dummy (with old vpn peer structure) under the new peer structure will fail to empty out the peers .. the dummy have to be on top - this is how it was with our firmware - and took us some hours of investigating

NeoRaider, 2016-01-22:
@viisauksena, that doesn't make any sense, the ordering of the peers and groups does not have any effect (in fact, Lua handles the table entries in arbitrary order).

viisauksena, 2016-01-22:
this was quiet a issue for us, while reordering produce double entry when keeping config - this doesnt have any effect on firstboot/factory .. the before state keeps old fastd entrys while sysupgrade, this commit empty out existing entrys : https://cccfr.de/git/viisauksena/firmware/commit/4f670b88b450bd9d5af3c52a9ee9ebd0cd28a922 
maybe youre right, and we have done something other wrong, actually i have no time to test this . (old firmware flashing, build ne FW with bogus update, flash new one with keeping setting, and same procedure with the updatet version)

NeoRaider, 2016-01-22:
@viisauksena, the mistake was that the "backbone" entry was outside of "groups" before.

viisauksena, 2016-01-22:
uff, what a little }  could do for trouble ... thx for looking at it

plumpudding, 2016-02-25:
1) Why not just add an option?
2) A warning when flashing a different site without discarding old settings?

A-Kasper, 2016-02-25:
Am 25.02.2016 um 06:40 schrieb plumpudding:
> 1) Why not just add an option?
> 2) A warning when flashing a different site without discarding old settings?

Maybe autodiscard if sitecode changes would help

viisauksena, 2016-02-25:
why? because maybe its intentional that you raise the number of peers ... you can add empty (oldnamed) groups , so they will be empty - to handle this in advance limit the options you have.

mmalte, 2016-02-25:
We had quite a lot of trouble with this so called feature in the past and it's really annoying to keep long gone peer groups in the site.conf. 

On the other hand I've never heard of someone using this to add custom fastd peers. 

I understand the idea behind the feature, but we should adjust it to the reality.

viisauksena, 2016-02-25:
my arguent was more, its overcontrolling if handled extra , therefore here is the flag documentation -
as the docuentation should make it really clear that old groups will not be overwritten (unless a similar empty group is added)

mmalte, 2016-02-25:
Well I see a lot of trouble caused by this behaviour and I'm absolutely sure that anybody who is able to set up an additional gateway is also able to create a simple persistent init.d skript which survives an Autoupdate.

tcatm, 2016-02-25:
Just to sum this up:

1. This can be "fixed" by better documentation on how groups are supposed to work.
2. Communities know which old peegroups may have existed.
3. It is encouraged to use a common peer group name (you can even nest other groups below it), like "backbone".
4. Sharing this peer group name with other communities will allow overwriting that group by any site.conf that also contains group named backbone.

Yes, this may cause some trouble for existing firmware. If required, we could introduce a temporary "reset-all-peer-groups" package for migrating to a common backbone group.

Is there anything I'm missing?

Adorfer, 2016-02-25:
I would suggest to create a database/lookuptable (perhaps in icvpn-meta) to keep track of the peergroups of different communities. 
(keeping track record of the ones which have been used in the past)

So it would be possible to maintain a local site.conf with empty peergroups for all "unwanted" groups which possibly could show up when people "reflash" routers with wrong setting ("keep existing setting"
(this would cause a site.conf of a few kB more empty groups, but it would eliminate the risk of domain-bridges a great deal.)

tcatm, 2016-02-25:
@Adorfer Please stay on topic. That's where an additional reset-peer-groups package could help.

Adorfer, 2016-02-25:
@tcatm 
 > Please stay on topic
I understood, that this issue should be resolved by better docoumentation, not by spoiling the "keep peergroups"-feature. 
So pushing out all unwanted peergroups (but keeping the ones not listed, presumingly "private" ones, wanted by the respective owners of those nodes) would need such a database.

viisauksena, 2016-02-25:
after thinking a bit more about this - beside better dosumentation ... 
there is one thing which could be done beside dokumentation ! 
that is, check if there are double fastd keys , because in this case fastd ignores both - and this is for nearby sure unwanted. This is how fastd works. And lead to the ugly situation that many peers are written in but in worst case, all are ignored, because all are doubled (which happens when people move change peer group name, and dont do a cleanup before, or adding an empty group with old peer group name)

A-Kasper, 2016-02-27:
Am 25.02.2016 um 15:27 schrieb Nils Schneider:
> Just to sum this up:
> 
>  1. This can be "fixed" by better documentation on how groups are
>     supposed to work.
>  2. Communities know which old peegroups may have existed.
> 

 [...]
> Is there anything I'm missing?

I don't see any problems in making correkt images for our own community.
The problems occure if people are moving to bochum and reflash their
device with bochum image without knowing, that they would have to clear
all settings. My suggestion was to delete old peers if communityname
changes...

Adorfer, 2016-02-28:
@A-Kasper 

> The problems occure if people are moving to bochum and reflash their device with bochum image without knowing, that they would have to clear all settings.

Lots of people like to reflash their routers to other communities without reentering the contact-info, Nodename, ssh-keys and lat/lon. 

So by consequence, all communities should document at least their current peer-groups in the api-file. 
By this you could at least integrate those peergroups as empty in your firmware, in order to reduce the risk of domain-bridges.

commit 3ccf7fdd (2020-05-10): gluon-mesh-vpn-fastd: clean up peers and groups on update

The 'preserve' flag can be used to mark a peer so it is not removed or
modified on upgrades. In addition, groups containing preserved peers are
not removed.

Fixes: #557
PR #2021 (2020-05-10): gluon-mesh-vpn-fastd: clean up peers and groups on update
The 'preserve' flag can be used to mark a peer so it is not removed or
modified on upgrades. In addition, groups containing preserved peers are
not removed.
    
Fixes: #557
closed 2020-05-10 by NeoRaider