freifunk-gluon/gluon #558: Luci doesn't finish line in /etc/dropbear/authorized_keys

MPW1412, 2015-11-14:
Hello,

I noticed, that Gluon doesn't finish the last line of /etc/dropbear/authorized_keys when ssh keys are added via the configuration mode.

If you pipe another key into that file later, you'll have the last key and the new one in the same line, which will make both of them unusuable. 

Bye
Matthias

tcatm, 2015-11-14:
The configmode SSH key field is written as-is to the file. If there's no newline at the end there won't be one in the file.

In order to reliably add a SSH key I'd suggest prepending the key to the beginning of the file:

    sed -i "1issh-rsa ....." /etc/dropbear/authorized_keys

jplitza, 2015-12-21:
If the sed expression scares you, you can also combine the two echos into one:
```
echo -e '\nssh-key ... user@host' >> /etc/dropbear/authorized_keys
```
As this is a problem of the kind "my manual modification didn't do enough sanitizing and thus broke what Gluon did before", I don't think this needs fixing in Gluon. @tcatm even documented his command line [in the Wiki](https://github.com/freifunk-gluon/gluon/wiki/Commandline-administration#add-ssh-key) for future reference. Thus, closing.

MPW1412, 2015-12-21:
I just think, that it might happen, that someone locks himself out of a system, where he has no physical access.

If you're like, don't use SSH if you don't know what you're doing, I'm fine with it.

NeoRaider, 2015-12-21:
I somewhat disagree. The file should always end with a newline, and fixing this in Gluon would be trivial.

commit 145cb4de (2016-02-05): gluon-luci-admin: ensure that the authorized_keys file always ends with a newline

Fixes #558
closed 2016-02-05 by NeoRaider