freifunk-gluon/gluon #559: rc.local is not executed at boot-time

RubenKelevra, 2015-11-15:
As far as I know it is common to execute /etc/rc.local at boot-time, we wanted to use this file to fix an issue with bridge-configurations ... so it should wait 60 seconds and add an interface to a bridge.

Nothing happens, so we installed a execution in /etc/init.d/network, this works, so it seems that /etc/rc.local is not executed at all.

We use master at commit 0b77bee3d7ce19696dec8f88922afadfe45a5fa8.

NeoRaider, 2015-11-15:
I can't reproduce this issue here, rc.local is run both in setup mode and normal operation. Could you post the command you're trying to run?

RubenKelevra, 2015-11-16:
'''
sleep 60

brctl addif br-client eth0.100
'''

As hotfix for #600

NeoRaider, 2015-11-16:
Well, you'll have to find out what is wrong yourself, there's nothing wrong with rc.local. Maybe procd doesn't like long-running scripts there and you'll have to background it?

"""
(
  sleep 60
  brctl addif br-client eth0.100
) &
"""
Which is obviouly really dumb anyways, as you could just use a hotplug.d script for such hacks instead of sleeping (or find out why netifd doesn't do this correctly).

closed 2015-11-16 by NeoRaider