freifunk-gluon/gluon #560: Issues with vlan-interfaces and client-bridge

RubenKelevra, 2015-11-15:
We tried to run x86-Gluon on a Siemens Futro S550. Everything works fine, but we want to route the Freifunk-Client-Interface back via VLAN-tagged packets to route it to the access-points, which should get Freifunk.

Now we added an VLAN-Interface like this:

config interface 'client_wan'
	option proto 'none'
	option ifname 'eth0.100'

And want to add this to the client-bridge:

config interface 'client'
	option type 'bridge'
	option macaddr '00:19:99:5f:ad:e8'
-	list ifname 'bat0'
+	list ifname 'bat0 eth0.100'
	option proto 'dhcpv6'
	option reqprefix 'no'
	option igmp_snooping '0'
	option peerdns '1'
	option sourcefilter '0'

Is this wrong, since if we reboot with this "ifname"-listing, the client-bridge disappear? We didn't find any useful tutorials for vlan-configurations on openwrt, so this was only a guess.

Using `brctl addif br-client eth0.100` after boot, seem to work ... so is this setup wrong?

Best regards

Ruben

jplitza, 2015-11-15:
I think `ifname` shouldn't be a list but a string. At least, that's what it is in my node:
```
	option ifname 'eth0.1 bat0 client-vpn'
```
And if it actually should be a list, your example should read
```
    list ifname 'bat0'
    list ifname 'eth0.100'
```

NeoRaider, 2015-11-15:
We're using a list since Chaos Calmer (both is supported by OpenWrt), so @jplitza's second example is correct.

closed 2015-11-15 by NeoRaider