freifunk-gluon/gluon #573: IBSS disabled wird ignoriert

A-Kasper, 2015-11-27:
Ich will unser Netz von IBSS auf 802.11s umstellen. 
Für einen Reibungslosen Übergang habe ich zunächst 802.11s ins Image eingefügt. Nach dem das Update durchgelaufen ist habe ich den Eintrag in der site.conf um disabled=true ergänzt. Obwohl es keine manuellen Einstellungen gab die diesen neuen default wert überschreiben würden, wird das IBSS Netz nicht deaktiviert und die disabled=true Einstellung ignoriert.

NeoRaider, 2015-11-27:
Die "disabled"-Option setzt den default. Die alte Einstellung hat immer Vorrang, da diese auch manuell gesetzt sein könnte. Soll IBSS deaktiviert werden, muss der Absatz ganz aus der site.conf entfernt werden.

closed 2015-11-27 by NeoRaider

A-Kasper, 2015-11-27:
wenn der alte default an war, kann man den nicht deaktivieren? mir ist
klar, dass hier der default gesetzt wird. ich möchte ibss nur deaktivieren,
aber manuell aktivierbar machen. dann einmal komplett ohne ibss flashen und
dann wieder mit deaktiviertem? das bringt doch auch nix?! der fw default
wert muss doch irgendwie gesetzt werden können solange keiner manuell etwas
ändert.
wir haben ja ein ähnlich unerwünschtes verhalten bei den peers. evtl.
bräuchten wir mal ein updatescrpt package, dass man einbinden kann und das
nachvdem update ausgeführt wird. dann könnte man in diesem mit uci
umkonfigurieren.

NeoRaider, 2015-11-27:
Bei vielen Optionen aus der site.conf setzen wir damit nur den initialen Wert, und können danach nicht mehr unterscheiden, ob das vom User gesetzt wurde oder nicht. Das ist bei praktisch allem, was irgendwie "aktivieren/deaktivieren" ist, der Fall (VPN, Mesh-on-LAN/WAN, WLANs, ...), und auch gewünscht so.

Wenn es 2 Zustände für eine Option gibt (z.B. IBSS an und aus), die beide erlaubt sind und von User konfiguriert werden können (im Expert Mode), sollte beim Update diese Option nicht verändert werden, egal ob der alte Zustand vom User gesetzt wurde oder nicht. Eine Option sollte nur beim Update geändert werden, wenn der alte Wert keine gültige Einstellung mehr ist. (Das ziehen wir natürlich nicht völlig konsequent durch in Gluon... aber wir versuchen, uns an der Regel zu orientieren).

Ich würde empfehlen, beim Wechsel auf 11s endgültig Abstand von IBSS zu nehmen und das nicht mehr aktivierbar zu machen. Solange IBSS noch gebraucht wird, würde ich vom Wechsel auf 11s absehen...

A-Kasper, 2015-11-27:
Am 27.11.2015 um 13:41 schrieb Matthias Schiffer:
> Bei vielen Optionen aus der site.conf setzen wir damit nur den initialen
> Wert, und können danach nicht mehr unterscheiden, ob das vom User
> gesetzt wurde oder nicht. Das ist bei praktisch allem, was [...]

Uns bricht diese policy immer und immer wieder das genickt und erschwert
die Netzwerkpflege. Eine Anregung wäre konsequent zwischen Default und
Userwerten zu unterscheiden und im Code die defaultwerte nur zu nutzen,
wenn keine userwerte vorliegen. Das würde auf der seinen Seite die
Experteneinstellungen erhalten, auf der anderen Seite aber nicht nicht
manuell konfigurierten Router (die allermeisten) entsprechend über die
admins ohne usereingriffe pflegen lassen. Das könnte man dann auch bei
den peers übernehmen: Setzt wer im lua einen Haken bei Custom Peers
würden die Default Peers einmalig kopiert und als userwerte vorliegen.
Würde man den Haken weg machen würden sie gelöscht und wieder nur die
defaultwerte genutzt.

Eines der Ziele von gluon war immer "Idiotensicher". Dafür wäre es gut,
wenn die configs weitreichend über die images gesteuert werden können,
damit die 95% User die einfach nur ein Default Image fahren sich nicht
kümmern müssen.

> Ich würde empfehlen, beim Wechsel auf 11s endgültig Abstand von IBSS zu
> nehmen und das nicht mehr aktivierbar zu machen. Solange IBSS noch
> gebraucht wird, würde ich vom Wechsel auf 11s absehen...

Das werde ich jetzt so machen. IBSS wollte ich nur unterstützen, damit
wer, der das update verpasst hat nochmal die Möglichkeit hat seine mesh
node zu retten. aber dann soll diese person halt eine alte firmware nehmen.