freifunk-gluon/gluon #576: mac hiding of clients - privacy against VDS-freifunk-style

viisauksena, 2015-11-28:
not sure if this is better in gluon or batman-adv 

in many cases people want to protect the original mac adresses of their clients .. so it could be possible for a router/node to fake a clients mac to another ... 
somehow mac-nat

if this is done in a defined way, you wont even loose roaming - because if some routers on a lonely isle alltogether use the same way of faking the mac itts persistent.

this will keep the big infrastructure in background with batman-adv intact ..
it only means that 1 or some routers under control by you will change clients mac - inside your node-mesh the roaming will work

in my mind are some settings where the protection of clients mac has a far higher importance than roaming through the whole net (which is mostly unrealistic at the end) 
idea came up, because actually you can track any user (mac) through ẃhole comunity networks, with the meshviewer data, like geo (etc) you will get a pretty good Vorratsdatenspeicherung-the-freifunk-way
whie i see layer2 meshing is like it is, i do not see why nodes could not change macs to protect real-mac if its desired

NeoRaider, 2015-11-28:
We've thought about this for a while, but haven't found a satisfactory solution for hiding the MAC addresses yet.

Roaming between any two points of the mesh is very important, because two nodes that are in physical proximity may not be in network topological proximity (they may not see each other except via VPN, but clients roaming between them may be very common). It is very important to us to adhere to the definition of an ESSID, which is: If two nodes announce the same ESSID, roaming is possible. Also, we'd like to keep the current state of not differentiating between VPN and WLAN mesh links, if possible.

Hiding the MAC is something we definitely won't implement as long as we use batman-adv. batman-adv couples clients and mesh too tightly, as the clients are directly bridged into the mesh.

When we have implemented a layer 3 mesh, we will use IP addresses instead of MAC addresses for routing; the MAC addresses will be used for roaming only. Any work towards hiding identification information should be developed for this scheme. Unfortunately, I don't think we can really hide this information at all, as we definitelly want to allow reaching all clients by all their IP addresses at any time. Even the IPv6 privacy extensions only add additional IP addresses, so there's still an address based on the MAC address. IPv6 NAT is a no-go.

Furthermore, in my opinion, hiding the MAC addresses isn't a responsibility of the mesh; rather, the clients should change their MAC addresses on a regular basis. A Freifunk network is inherently the most insecure network possible. Anyone can create a node, anyone can track everything, and with a little effort, anyone can read and modify *all traffic*, both metadata and payload. This is an intrinsic property of such open networks.

So instead if a technical problem, this is a social one: Tell everyone that Freifunk is not secure, and the users should always use E2E encryption, change their MACs, etc. I know this is not the best promotion for Freifunk, but telling people that Freifunk is secure in any way is an outright lie in my opinion.

viisauksena, 2016-01-03:
i agree with most terms .. but would also love to see a solution which allow special nodes (or group) of nodes scrumble their macs ... their is at least a minimum on technical responsibilty you have by building infrastructure and risk privacy of people using it (especially when they do "their best").
again, i agree with you - but i will look forward in mac-natting with batman - i also spoke with the batman-adv people over this issue - its not on their top list _ but on mine

viisauksena, 2016-02-09:
adding this proof of concept script of tracking mac addresses 
https://cccfr.de/git/viisauksena/toolbox/blob/master/tracker_poc.sh

RubenKelevra, 2017-03-27:
Well, as long as we use Batman a scambling would not just hurt roaming but completly destroy it. So devices would stay in the network but since the MAC-Adress changes the possibility is high that the network has troubles to keep the connections open, also excessively adding unneeded management traffic, because all roamings create new mac addresses.

Furthermore some devices renew their leases on roam some not, surely the get a new IP if they ask for a renew, because they now ask with a different mac-address.

All in all scrambling the mac-address doesn't work for us. This must be done on the devices or in the routing protocol itself and must be stable, like calculating a sha512 from the mac and the date and generating the mac for the network from this, which can be redone on each node network wide.

mweinelt, 2020-03-02:
Thanks everyone for your concern, but with the current routing protocols in place this bug is not actionable.

closed 2020-03-02 by mweinelt