freifunk-gluon/gluon #577: Nach 'uci set system...hostname...' reicht 'commit system' nicht aus, reboot nötig?

KarlaB, 2015-11-30:
(english description at the bottom)

Aloha rundum,

wenn ich auf einem TL-WR841 v9 mit ffmstable-1.10 / gluon-v2014.2-1011-gc16522b via SSH den hostname ändere, 

`uci set system.@system[0].hostname='newname'`
`uci commit system`

dann dauert es nur kurz und ich kann in der map oder der status page die Änderung sehen.

Mache ich das gleiche auf einem Gerät der gleichen Hardwareversion aber mit ffmdev-3.83 / gluon-jenkins-ffm_gluon-dev-83 (hier allerdings login nicht per passwort sondern per public key), dann ändert sich der hostname erst nach einem uci reboot oder Strom aus/Strom an.

Liegt das an einer der Änderungen im gluon?

Merci und schöne Grüße
Nadja

PS: Gleiches scheint für die Änderung der location zu gelten.

Und der Benutzername im SSH-Prompt bleibt bei diesem Beispiel anscheinend bei beiden gluon-Versionen auch bis zum Neustart noch der alte. Aus- und Einloggen allein hilft auch hier nicht ...

ENGLISH DESCRIPTION:

Aloha ahiahi,

when changing the hostname on an 841 via SSH with

`uci set system.@system[0].hostname='newname'`
`uci commit system`

there is a difference between 
ffmstable-1.10 / gluon-v2014.2-1011-gc16522b
and 
ffmdev-3.83 / gluon-jenkins-ffm_gluon-dev-83

On the first version, the changes are visible in the map or on the status page after a few minutes while the new (dev) version needs a reboot before the changes take effect.

Is the reason in one of the changes to gluon?

Thanks a lot
Nadja

NeoRaider, 2015-11-30:
Well, UCI changes only the config files, usually all config changes require restarting the corresponding services or rebooting.

This particular change is an effect of changing how the announced data is generated. It is now collected in a persistent process (gluon-announced/respondd) with a persistent Lua context, and this persistent UCI context. To make respondd refresh the UCI config, just restart it using `/etc/init.d/gluon-announced restart`.

closed 2015-11-30 by NeoRaider

KarlaB, 2015-11-30:
Thank you, that sounds as if one could avoid to restart the hole device - but in my case there is no such "/etc/init.d/gluon-announced":

`root@xyz:/etc/init.d# ls`
`alfred             fastd              led                sysntpd`
`boot               firewall           log                system`
`cron               gluon-ebtables     micrond            telnet`
`dnsmasq            gluon-radvd        network            uhttpd`
`done               gluon-wan-dnsmasq  sysctl             umount`
`dropbear           haveged            sysfixtime`

Did "gluon-announced" get lost by accident or is there another way?

Btw: Could it be possible and helpful to execute `/etc/init.d/gluon-announced restart` automatically when `commit` is executed? If not many tutorial should be adapted ...

Thanks again
Nadja

NeoRaider, 2015-11-30:
`uci commit` just writes the config files, nothing more. uci is just a way to store configuration files, there's no magic involved.

Hmm, indeed, I just realized that we don't start announced using a normal init script, but using /etc/hotplug.d/iface/10-gluon-announced... I think something like `sh -c 'source /etc/hotplug.d/iface/10-gluon-announced; restart_announced'` might work, but I can't test it at the moment. We should probably pull out that command into a new script in the next Gluon release.

KarlaB, 2015-11-30:
Great, `sh -c 'source /etc/hotplug.d/iface/10-gluon-announced; restart_announced'` worked well!

So maybe you want to keep this issue open?

Have a good night
Nadja

Little-Ben, 2015-11-30:
I had that problem too, with the autoupdater branch. at least we have a 'workaround' now. thanks @NeoRaider