freifunk-gluon/gluon #585: Problems after autoupdate

A-Kasper, 2015-12-07:
I've got many reports (about 12), that nodes are dead after autoupdate. I can't say in which cases this happens. different hardware is affected. All devices are working fine after a powercycle. 
I just had one case with my own routers. I was in hurry and needed to make the network work again, so there was no time for deeper investigations. But I can report, that I was able to connect to the router via LAN IP and SSH. It still answered, so it wasn't frozen. Seems like something doesn't always come up after autoupdate. 
Can someone confirm this issue?

NeoRaider, 2015-12-07:
@A-Kasper, did the version installed before running the updater have IBSS and 11s enabled at the same time?

Some logs (logread) would be nice.

jplitza, 2015-12-07:
Also, which version did they have before, did the upgrade succeed and if so, which version did  they have afterwards? What were the models affected? I'm suspecting the autoupdate filling the RAM of some of the smaller models and thereby killing them.

A-Kasper, 2015-12-07:
The problem occured with different hardware. I remember 841 v9, 841 v10,
CPE210 and Futro offloader. I don't got any logs. But don't think, that it
is a memory problem. The update is successfull and the Futro shouldn't got
memory issues.

The old firmware were 2015.1.2 or 2015.2. In case of 2015.2 both mesh
Networks were enabled and mesh disabled in the new fw.

A-Kasper, 2015-12-07:
So... schon wieder. Ich muss korrigieren: ssh ist auch Tot. Über Konsole kam man nur dran. Die Kisten die beim letzten update weggeknallt sind sind aber alle für mich nicht greifbar. 
Eine Gemeinsamkeit scheint zu sein, dass Mesh on LAN (oder Mesh on LAN?) angeschaltet ist. Jedenfalls scheinen Router betroffen zu sein, die in irgendeiner Form Offloader spielen.

Delta1977, 2015-12-07:
ich hatte ich schon so etwas. immer nur Nodes mit VPN-Uplink. Die erste Gluon die diese Nodes gesehen haben waren ein  Masterbuil vor ca 4 Wochen und  zwischendurch immer mal wieder auf einen aktuellen Build geflasht. Die Nodes waren im 802.11s mode, IBSS hatten die nie gesehen.

Manchmal half ein Powercycle, bei 2en brachte nur ein tftp recovery was

A-Kasper, 2015-12-07:
Am 07.12.2015 um 17:47 schrieb Delta1977:
> ich hatte ich schon so etwas. immer nur Nodes mit VPN-Uplink. Die erste
> Gluon die diese Nodes gesehen haben waren ein Masterbuil vor ca 4 Wochen
> und zwischendurch immer mal wieder auf einen [...]

Beim aktuellen Update von gestern Nacht bei uns kann ich das leider
bestätigen. Bei uns sind dieses mal auch welche dabei die auch nach
Powercycle nicht zu reanimieren waren.

bittorf, 2015-12-08:
@A-Kasper waren das alles ath9k-Geräte?

A-Kasper, 2015-12-08:
nein. die Futros haaben kein wlan.

NeoRaider, 2015-12-08:
Unreliable upgrades are certainly a severe issue, but without logs, I don't really know where to start. I haven't seen any such issues here, but maybe the network in Lübeck is too small. @A-Kasper, how many nodes and clients does your network have?

If the upgrade was from 2015.1.x to 2015.2, it's an issue in 2015.1.x, so knowing the exact versions the upgrades failed with would be very important.

A-Kasper, 2015-12-09:
Hi Matthias,

Am 08.12.2015 um 23:21 schrieb Matthias Schiffer:
> Unreliable upgrades are certainly a severe issue, but without logs, I
> don't really know where to start. I haven't seen any such issues here,
> but maybe the network in Lübeck is too small. @A-Kasper [...]

I'm very sorry that I'm stil unable to collect more facts. This evening
is our Freifunk-Treff. I'll ask for more information. Otherwise I'll try
to reproduce the issue at home with a Futro where I've access to the
console in case of error.
I've found some ath: skbuff alloc of size 1926 failed Errors in my
ubiquiti bullet with this firmware. google says the memory is full.
802.11s is enabled in this firmware. maybe the autoupdater fails because
of the full memory? I'll try to get more information as soon as possible.

RubenKelevra, 2015-12-14:
Try something like this for update:

( sleep 500; reboot -f )&
sysupgrade /tmp/gluon.img 

This might solve the reboot-hang.

NeoRaider, 2015-12-20:
As this is a OOM issue, solving #416 would probably help.

bittorf, 2015-12-20:
maybe it is a good idea during sysupgrade to activate panic on oom:
```
/proc/sys/vm/panic_on_oom = 2
/proc/sys/kernel/panic_on_oops = 1
/proc/sys/kernel/panic = 10
```

jplitza, 2015-12-20:
While you talk only about panic on OOM, you mention two other sysctls there without any explanation.

* `sys.vm.panic_on_oom` is actually always set to 1, not only during autoupdate. I couldn't find out what the "2" you want to write there does differently.
* `sys.kernel.panic` is actually always set to 3, so you would only extend the delay of reboot.
* `sys.kernel.panic_on_oops` might or might not be a good idea. Probably could have a positive effect to always set this to 1.

bittorf, 2015-12-20:
@jplitza thanks for not beeing lazy as me. 
here is the full doc:
https://www.kernel.org/doc/Documentation/sysctl/vm.txt

so: panic_on_oom=2 means:
```
If this is set to 1, the kernel panics when out-of-memory happens.
However, if a process limits using nodes by mempolicy/cpusets,
and those nodes become memory exhaustion status, one process
may be killed by oom-killer. No panic occurs in this case.
Because other nodes' memory may be free. This means system total status
may be not fatal yet.

If this is set to 2, the kernel panics compulsorily even on the
above-mentioned. Even oom happens under memory cgroup, the whole
system panics.
```
sys.kernel.panic is NOT always 3. who should set this?

NeoRaider, 2015-12-20:
@bittorf: `git grep -w panic: package/base-files/files/etc/sysctl.conf:kernel.panic=3`

bittorf, 2015-12-21:
@NeoRaider thanks

corvusmo, 2016-01-27:
We have still the problem, that a TL-841 (v9) does not survive autoupdate and has to be recovered via tftp. The last time it happened on the current master (update from 4ed5b856688d4e13abb1ae84b01ab5cce991ef06 to 2223ccd94c7d1fa3e563d2c68fc116d8fca4a663, build by @aeickho). Both 802.11s and IBSS are running, the network (Freifunk Göttingen) has about 600 nodes online.

Could it be interesting to try with 802.11s disabled?

rotanid, 2016-01-27:
have you read the other comments? especially "Unreliable upgrades are certainly a severe issue, but without logs, I don't really know where to start."
additionally, the improvements to the update process are done but haven't been merged in the master branch so far, i think you have to be patient, (master isn't recommended for production anyway)

NeoRaider, 2016-01-28:
@corvusmo, as @rotanid mentioned, there are a few improvements waiting to be merged (to be exact, the c-respondd branches of the gluon and the packages repos), which will reduce memory consumption and thus make the nodes and the updates in particular more reliable.

As mentioned in the preliminary release notes for Gluon 2016.1, using 11s and IBSS at the same time is strongly discouraged, as smaller nodes (as the TL-WR841N) can't handle the doubled load. Unreliable updates are expected in such configurations. **I repeat: If you have enabled both 11s and IBSS, your updates will be unreliable.**

viisauksena, 2016-01-29:
wir haben ein paar probleme mit Knoten die wegen ein anderen firmware (einen batzen ath rx Fehler ausgegeben haben) und bestimmt auch am Speicher/Load Limit waren. Die wurde mittels ssh zum update gebracht, es waren allesamt Meshende Knoten. Diese Knoten haben dann zwar geupdatet sind aber nicht mehr durch powercycle, oder den config-mode, oder den rescuemode zu erreichen gewesen. Die sind in einem 5 sekunden rebootloop - sprechen aber noch tftp (mangels laufender tftp kann ich nicht testen ob die daraus heil wieder zu retten sind).
das betrifft von 20 bisher geflashten 841v9 und v10 und einigen cpe 210 - bei uns immerhin 4 der 841er. 
Wirklich sagen woran das liegt lässt sich vermutlich auch nach dem "retten" nicht mehr. Es kann auch sein das die schlicht ein fehlerhaftes update versucht haben einzuspielen, oder während dessen schlicht abgesemmelt sind, so dass sie nicht viel mehr als die bootpartition geschrieben haben. Eigentlich sollte sowas nicht passieren können.

A-Kasper, 2016-01-29:
kann man in den autoupdater nicht irgendwas integrieren dass er zunächst
alle dienste beendet und speicher freiräumt bevor er die eigentliche fw
runterlädt und updated? Macht der autoupdater einen md5 check?

jplitza, 2016-01-29:
Genau das Beenden von Diensten vor dem Runterladen sowie das Neustarten nach einem fehlgeschlagenen Update ist in freifunk-gluon/packages#128 vorbereitet worden. Und natürlich führt der Autoupdater einen Checksum-Test durch, sonst wäre das ganze Manifest samt Signaturen ja sinnlos.

NeoRaider, 2016-02-02:
0005b3b5a2e7ceddedaff4196104d185c76d2923 should improve the situation significantly. Feedback welcome.

mmalte, 2016-02-03:
Ich habe die Erfahrung gemacht dass überlastete Knoten, wie ein 841v9 in einem 1.200 Knoten mesh das Autoupdate zigfach nicht schaffen, aber neu booten.

Wenn ich zuvor das client wifi abgeschaltet habe klappte es stets auf Anhieb. Zudem wird der download des Updates dadurch beschleunigt.

Möglicherweise wäre die auch für den Autoupdater eine hilfreiche Ergänzung.

jplitza, 2016-02-03:
Insbesondere in Kombination mit #582 ist das keine gute Idee, da das massiv den hauptsächlichen Sinn des Knotens einschränkt.

A-Kasper, 2016-02-03:
Es wäre vermutlich gar nicht so verkehrt das clientnetz im
https://github.com/freifunk-gluon/gluon/commit/0005b3b5a2e7ceddedaff4196104d185c76d2923
auch noch abzustellen. Schaden tut es imho nicht. Der Router bootet eh 10
sekunden später neu, die Clients verlieren also eh ihre Verbindung. Wenn es
hilft... why not!?

NeoRaider, 2016-02-03:
@A-Kasper, in 0005b3b wurde sehr darauf geachtet, die Haupt-Funktionen des Knotens nicht zu beeinträchtigen, da beim Download des Images sehr viel schief gehen kann, und somit völlig unklar ist, ob und wann danach tatsächlich ein Update passiert. In #582 wurde ja schließlich gesehen, dass der Download des Images über mehrere Stunden hängen kann.

Was denkbar und vielleicht sinnvoll wäre, wäre, das Netzwerk *nach* dem Download des Images völlig herunterzufahren, damit beim Flashen auf keinen Fall die Ressourcen ausgehen.

A-Kasper, 2016-02-03:
Das meinte ich auch so. Vorher würde keinen Sinn machen. Wenn da vorher
etwas schief geht failed ja die Prüfsumme... Es reicht also für die
Stabilität das hinterher zu machen

mmalte, 2016-02-03:
Wenn man das Client Netz abschaltet ist zumindest bei den ~30 Knoten die ich so aktualisiert habe nie einer hängen geblieben. Die Nutzer nachdem anhand des Manifest festgestellt wurde dass ein Update durchgeführt wird zu kicken hat zudem den Vorteil dass der Knoten schneller wieder einsatzbereit ist.

jplitza, 2016-02-04:
Um das nochmal klar zu stellen, der Update-Vorgang ist momentan wie folgt:

1. Manifest herunterladen
2. Manifest verifizieren
3. Feststellen, ob es eine neue Version gibt
4. Dienste herunterfahren
5. Image herunterladen
6. Image verifizieren
7. sysupgrade aufrufen

Schritt 4 kommt vor Schritt 5, weil der Download des Images in den RAM geschieht und dementsprechend bereits hohe Anforderungen an den freien RAM hat. Schritt 5 ist auch der, der potentiell lange dauert.

Ihr schlagt nun vor, zwischen Schritt 6 und Schritt 7 das Client-Netz auszumachen, ja? Hat der Download des Images mit eingeschaltetem Client-Netz funktioniert?

mmalte, 2016-02-04:
Das trifft es recht gut, hier nochmal kurz warum ich es nach meiner Erfahrung für einen sehr guten Plan halte:
Ich habe in diversen Flüchtingsunterkünften und bei eingen befreundeten Gastronomen per ssh updates eingespielt, insgesamt bei gut 30 Geräten. Das ganze per autoupdate mit manuell reduzierter Signaturzahl um nicht die gesamte Domäne zu aktualisieren.

Zu diesem Zeitpunkt hatten wir im Netz über 1.200 Knoten laut originator List. Vielen der Knoten hatten über 50 Clients.

Ich habe zunächst versucht es mittels:

```
uci set autoupdater.stable.good_signatures=3
autoupdater

```
Dabei habe ich nur bei jedem 10. Versuch das update geschafft und es dauerte ewig. Selbst Archer C5 sind beim update vereinzelt gescheitert wenn sie viele Clients hatten. Da auch schon der Download Prozess quälend langsam war habe ich das client Netz abgeschaltet:

```
uci set autoupdater.stable.good_signatures=3
wireless.client_radio1.disabled=1
wifi
autoupdater
``

So wurde Update stets zügig geladen und hat zu 100% beim ersten Versuch funktioniert. Bei den 841v10 die ich  noch nicht per Autoupdater versorgt habe bin ich analog mit zusätzlichem drop cache und sysupgrade vorgegangen und war auch ausgesprochen erfolgreich.

Unsere Domäne ist inzwischen in Segmente mit ~200 Knoten zerlegt, ich kann also nicht mehr unter diesen extremen Bedingungen testen.

Zusammenfassend halte ich es für viel besser einen für kurze Zeit für Clients überhaupt nicht nutzbaren knoten zu haben, als einen für längere Zeit nur sehr eingeschränkt nutzbaren Knoten zu haben.

jplitza, 2016-02-04:
Genau das hatte ich befürchtet. Was du schilderst war eine Deaktivierung des Client-WLANs *bevor* das Image heruntergeladen wurde (oder der Autoupdater überhaupt irgendwas getan hat). Was ich schilderte war eine Deaktivierung *nachdem* das Image heruntergeladen wurde. Es ist also unklar, ob diese spätere Deaktivierung auch hilft.

Das Client-WLAN vor dem Download zu deaktivieren halte ich für unmöglich, so lange #582 nicht resolved wurde.

mmalte, 2016-02-04:
Beim Autoupdater würde ich es dazwischen schieben.

Ab dem Moment in dem der Freifunk Router mit dem Download einer Firmware beginnt sollte das seine Hauptaufgabe sein. Clients sollte nun keiner Rolle mehr spielen

Während des Downloads ist er nicht mehr sinnvoll einzusetzten. Selbst bei enorm schlechter Leitung würde ich sagen: Lieber fünf Minuten früher das Client Netz abgeschaltet und dafür auch früher wieder online.

Das ganze natürlich immer nur wenn das manifest bestätigt wurde.

A-Kasper, 2016-02-04:
Ich gebe JPL da völlig recht... der Download ist überhaupt kein
Stabilitätskritischer Vorgang. Dann klappt er halt 10 mal nicht. Who
cares? Um 4 Uhr Nachts wird es ihm schon gelingen...

Es geht hier bei dem Problemen um Stabilität. Das Autoupdate muss
zuverlässig laufen. Wenn man alles runterfährt nachdem das image im
Speicher liegt reicht das dicke...

Ich kann übrigens vermelden, dass unser letztes update stabil
durchgelaufen ist. Zumindest ist mir kein Ausfall bekannt geworden. In
unserem Fall war es vermutlich tatsächlich der hohen Speicherlast durch
die zwischenzeitige Nutzung beider Meshnetze, die wir aktiviert hatten
um einen Reibungslosen Umzug zu ermöglichen. Ich gehe davon aus, dass
der neue Fix ausreichend sein sollte das update stabil zu fahren.

mmalte, 2016-02-04:
Mir ist es stets lieber wenn das Client Netz aus ist als wenn es schlecht Funktioniert, aber bei einem brauchbaren Uplink geht es hier ja auch nur um 1-2 Sekunden Unterschied. Nur bei echten Problemknoten macht es einen Unterschied ob man das Client Netz vor oder nach dem download abschaltet.

Ein Abschalten nach dem Download ist auch für mich hinreichend, vor dem Download wäre mir jedoch lieber.

Adorfer, 2016-02-05:
Ich vermag zu sagen, dass ich mit von dem aktuellen master (inkl. 0005b3b) rund ein Dutzend Update-Tests gefahren habe, incl "Tief im Mesh und während des updates des Knotens den Uplink abgeschaltet". 
Ergebnis: Knoten sind in den wget-Timeout gelaufen, haben die Handvoll abgeschalteten Dienste wieder hochgefahren und eine Stunde später wieder probiert. 

Des weiteren habe ich mir mittels eines Hirnknotens Anfang Januar ein WorstCase-Szenario gebaut: Updateschleife (Details erspare ich mir hier). Faktisch haben rund 30 Knoten so viele Updates geschoben, wie es nur ging innerhalb von rund 19 Stunden. Einige haben das mehrmals in der Stunde gemacht. (es war ein Build aus dem Master der ersten Januar-Woche, noch ohne "crespondd"): 
Ergebnis: Keine Toten. Nur ein Node mit einem schlechten POE-Netzteil/Kabelüberlänge war weggstürzt und brauchte einen Powercycle, kam dann aber wieder hoch. 
(Mir ist klar, dass das nicht gut für die Flash-Chips war....)

Lange Rede, kurzer Sinn: Zumindest von der Funktionalität scheint der Autoupdater meinen Ansprüchen derzeit zu genügen. Das war mit AA viel schlimmer.

NeoRaider, 2016-02-05:
Sound good, closing for now. If anything new comes up, we can reopen this bug.

closed 2016-02-05 by NeoRaider