freifunk-gluon/gluon #588: no kernel modules for GLUON_TARGET=ar71xx-nand

aeickho, 2015-12-08:
with:
make GLUON_TARGET=ar71xx-nand 

thera a no files on ./output/modules

NeoRaider, 2015-12-08:
I can't reproduce this issue here, what's the exact command you're building with? Please provide logs.

aeickho, 2015-12-08:
make GLUON_TARGET=ar71xx-generic GLUON_BRANCH=experimental GLUON_RELEASE="m_ae_201512080141"
GLUON_TARGET=ar71xx-nand GLUON_BRANCH=experimental GLUON_RELEASE="m_ae_201512080141"

https://gist.github.com/aeickho/215efd156d003341f6be
? Line 12342

NeoRaider, 2015-12-08:
The log looks fine, the module directory is cleaned in l.12342, in l.12348, the new modules are copied, and in l.12349 the package index for the new module directory is created. There are no unexpected failure messages there, and the package indexing is working, so I think it is working correctly...

aeickho, 2015-12-08:
sorry, neoraider ... it's absolutely  correct  
I was looking for ar71xx-nand and not for ar71xx/nand  

:(

closed 2015-12-08 by aeickho